edgexfoundry/device-onvif-camera #2: Device Discovery should use EndpointReference for uniqueness

ajcasagrande, 2022-02-16:
**Observation**
We have discovered that our `Tapo C200` cameras all return the same serial number when queried via `GetDeviceInformation`.
We have also learned that the `happytime-multi-onvif-server` sets all camera serial numbers to `123456` by default.

**Problem**
Only 1 device of each type `Tapo C200` and `happytime` can be added to EdgeX due to name conflicts.

**Proposed Solution**

Use the `Endpoint Reference` as specified in the ONVIF spec. This is available during WS-Discovery:

> 
> **7.3.1 Endpoint reference**
> A device or an endpoint that takes the client role should use a URN:UUID [RFC4122] as the address property
> of its endpoint reference.
> 
> The device or an endpoint that takes the client role shall use a **stable, globally unique identifier that is constant
> across network interfaces** as part of its endpoint reference property. The combination of an wsadis:Address
> and wsadis:ReferenceProperties provide a stable and globally-unique identifier.

[Source](https://www.onvif.org/specs/core/ONVIF-Core-Specification.pdf)

We have indeed confirmed that even when the device IP address changes, the `EndpointReference` stays the same, and is unique to all of our cameras.

cloudxxx8, 2022-02-21:
@ajcasagrande do you agree with Bruce's solution?

weichou1229, 2022-02-23:
@ajcasagrande 

I can get the EndpointReference:EndpointReference from probe message as below two camera:

**TAPO C200**
```
<wsdd:ProbeMatch>
  <wsa:EndpointReference>
    <wsa:Address>uuid:3fa1fe68-b915-4053-a3e1-c006c3afec0e</wsa:Address>
     <wsa:ReferenceProperties></wsa:ReferenceProperties><wsa:PortType>ttl</wsa:PortType>
  </wsa:EndpointReference>
  <wsdd:XAddrs>http://192.168.12.128:2020/onvif/device_service</wsdd:XAddrs>
  ...
</wsdd:ProbeMatch>
```

**HIKVISION**
```
<d:ProbeMatch>
 <wsadis:EndpointReference>
    <wsadis:Address>urn:uuid:cea94000-fb96-11b3-8260-686dbc5cb15d</wsadis:Address>
 </wsadis:EndpointReference>
 <d:XAddrs>http://192.168.12.123/onvif/device_service</d:XAddrs>
 ...
</d:ProbeMatch>
```

So we need to extract the uuid value `3fa1fe68-b915-4053-a3e1-c006c3afec0e` and `cea94000-fb96-11b3-8260-686dbc5cb15d` to replace the serial number.  Is that correct?

PR #7 (2022-03-01): feat: Replace the serial number with endpointReference address UUID
Close #2

## PR Checklist
Please check if your PR fulfills the following requirements:

- [ ] Tests for the changes have been added (for bug fixes / features) **not impact the teat**
- [x] Docs have been added / updated (for bug fixes / features)

**If your build fails** due to your commit message not passing the build checks, please review the guidelines here: https://github.com/edgexfoundry/edgex-examples/blob/master/.github/CONTRIBUTING.md.

## What is the current behavior?

The serial number is the same for the different T [... cut]
commit 36f711c3 (2022-03-01): feat: Replace the serial number with endpointReference address

Replace the serial number with EndpointReference Address as the part of device name

Close #2
closed 2022-03-01 by cloudxxx8