edgexfoundry/device-onvif-camera #22: Startup delay/blocking when missing credentials

ajcasagrande, 2022-03-29:
On startup, the service attempts to create a device client for each registered device. This causes it to lookup the camera credentials secrets. When the secrets do not exist, it waits up to 2 minutes for them, for each device. This all happens serially and delays the starting of the service.

```
level=INFO ts=2022-03-29T00:39:55.184382784Z app=device-onvif-camera source=driver.go:86 msg="Initializing device client for '3fa1fe68-b915-4053-a3e1-1027f5ea88f4' camera"
level=WARN ts=2022-03-29T00:39:55.184402418Z app=device-onvif-camera source=driver.go:283 msg="Unable to retrieve camera credentials from SecretProvider at path '3fa1fe68-b915-4053-a3e1-1027f5ea88f4': Error, path (3fa1fe68-b915-4053-a3e1-1027f5ea88f4) doesn't exist in secret store. Retrying for 1m59.999998509s"
level=WARN ts=2022-03-29T00:39:56.184607352Z app=device-onvif-camera source=driver.go:283 msg="Unable to retrieve camera credentials from SecretProvider at path '3fa1fe68-b915-4053-a3e1-1027f5ea88f4': Error, path (3fa1fe68-b915-4053-a3e1-1027f5ea88f4) doesn't exist in secret store. Retrying for 1m58.999817495s"
...
level=ERROR ts=2022-03-29T00:41:55.234543205Z app=device-onvif-camera source=driver.go:90 msg="failed to initial device client for '3fa1fe68-b915-4053-a3e1-1027f5ea88f4' camera, skipping this device."
```

https://github.com/edgexfoundry-holding/device-onvif-camera/blob/27426708967fb713f3ef9396d946112ad1b0b7a6/internal/driver/driver.go#L80-L91

weichou1229, 2022-04-18:
The device service tries to get the credential if the camera required authentication.
Here's the function to get the credential
https://github.com/edgexfoundry-holding/device-onvif-camera/blob/c64c8ceb32d9e245c546dc7983f6cd9184b271bd/internal/driver/driver.go#L265-L274
The user can decrease CredentialsRetryTime by overriding the env DRIVER_CREDENTIALSRETRYTIME.

ajcasagrande, 2022-04-22:
@weichou1229 I think the problem is that on startup this is not done asynchronously. Thus one camera with missing credentials will hold up the N number of cameras with valid credentials.

PR #113 (2022-07-21): MAC Discover Authentication + Device Name Consistency + Discover Documentation
## PR Checklist
Please check if your PR fulfills the following requirements:

- [X] Tests for the changes have been added (for bug fixes / features)
- [X] Docs have been added / updated (for bug fixes / features)

**If your build fails** due to your commit message not passing the build checks, please review the guidelines here: https://github.com/edgexfoundry/edgex-examples/blob/master/.github/CONTRIBUTING.md.

## What is the current behavior?

## Issue Number:
Closes #86 
Closes #83 
Closes #22

## What is the new  [... cut]
closed 2022-07-21 by ajcasagrande