edgexfoundry/device-onvif-camera #24: Geo vision/Honey well/Bosch camera- Set scopes API not working correctly

vyshali-chitikeshi, 2022-04-04:
Summary of the problem:
Set scope API deleting all configurable scopes instead of updating scope item. 

Issue reproducible on below camera models:
Geo Vision GV-BX8700
Bosch DINION 6000HD
Honeywell HC30WB5R1

Repro steps:
1) Execute get scopes rest api to retrieve existing scopes
2) If there are no existing 'configurable' scopes, then add  2 new scopes with 'Add scopes' API
3) Execute get scopes api and check 2 new configurable scopes have added
4) Execute set scopes api to update all configurable scopes with updated scope item.  Attached is the postman screenshot showing how we are executing 'set scopes' api.

Expected result: all configurable scopes updated correctly with desired scope item mentioned in 'set scopes' command
Actual result: Deletes all configurable scopes

Note: Not getting any error after executing ' set scopes' api, it gives 200/success message.

vyshali-chitikeshi, 2022-04-04:
[image]

weichou1229, 2022-04-18:
According to the spec https://www.onvif.org/ver10/device/wsdl/devicemgmt.wsdl#op.SetScopes
The device service treats the parameter`Scopes - unbounded; [anyURI]` as an array.
Please try the following JSON body:
```
curl --location --request PUT 'http://0.0.0.0:59882/api/v2/device/name/Camera003/Scopes' \
--header 'Content-Type: application/json' \
--data-raw '{
    "Scopes":{
        "Scopes":[
            "http//:123"
        ]
    }
}'
```

vyshali-chitikeshi, 2022-04-21:
Followed the recommendation, set scopes still not updating all confiugratble scopes as expected. Tested on geo vision camera.
One more question, in the test metric, for tapo camera 'add scopes' marked as 'x' and set scope marked as 'o', to execute set scopes first we need to add configurable scope using 'add scope', if 'add scope' is not working on tapo camera how you guys executed set scopes command?

brianointel, 2022-04-22:
verified with the following in postman:

{
    "Scopes":{
        "Scopes":[
            "http//:testscope"
        ]
    }
}

getScopes returned:

{
    "apiVersion": "v2",
    "statusCode": 200,
    "event": {
        "apiVersion": "v2",
        "id": "c43f741b-2a88-416d-b03e-e29176180464",
        "deviceName": "Camera001",
        "profileName": "onvif-camera",
        "sourceName": "Scopes",
        "origin": 1650650852185086200,
        "readings": [
            {
                "id": "ee3d80df-f160-4201-ad79-a05f3d2865d6",
                "origin": 1650650852185086200,
                "deviceName": "Camera001",
                "resourceName": "Scopes",
                "profileName": "onvif-camera",
                "valueType": "Object",
                "objectValue": {
                    "Scopes": [
                        {
                            "ScopeDef": "Fixed",
                            "ScopeItem": "onvif://www.onvif.org/type/Network_Video_Transmitter"
                        },
                        {
                            "ScopeDef": "Fixed",
                            "ScopeItem": "onvif://www.onvif.org/name/Bosch"
                        },
                        {
                            "ScopeDef": "Fixed",
                            "ScopeItem": "onvif://www.onvif.org/hardware/DINION_IP_starlight_6000_HD"
                        },
                        {
                            "ScopeDef": "Fixed",
                            "ScopeItem": "onvif://www.onvif.org/Profile/Streaming"
                        },
                        {
                            "ScopeDef": "Configurable",
                            "ScopeItem": "http//:testscope"
                        },
                        {
                            "ScopeDef": "Fixed",
                            "ScopeItem": "onvif://www.onvif.org/Profile/G"
                        },
                        {
                            "ScopeDef": "Fixed",
                            "ScopeItem": "onvif://www.onvif.org/Profile/T"
                        }
                    ]
                }
            }
        ]
    }
}

weichou1229, 2022-06-13:
Close this issue because it's not the device-onvif-camera's bug. 
Since some cameras do not support the ONVIF very well and we can verify the camera's capability by SOAP API, please reopen this issue if you have further questions.

closed 2022-06-13 by weichou1229