edgexfoundry/device-onvif-camera #25: Geo vision/Honeywell/Bosch camera- Set DNS API not working correctly

vyshali-chitikeshi, 2022-04-04:
Summary of the problem:
Set DNS API not updating camera DNS settings correctly.

Issue reproducible on below camera models:
Geo Vision GV-BX8700
Honeywell HC30WB5R1
Able to enable/disable DHCP on Geo vision and Honeywell cameras, however not able to update IP address with 'DNS Manual"
Not getting any error, but IP address not updated when verified using Get DNS api.

Bosch DINION 6000HD
Enabling/disabling of DHCP not working

Repro steps:
1. Execute get DNS  rest api to retrieve existing DNS configuration
2. Enable or disable DHCP using set DNS API
3. Execute Get DNS api again to verify DNS settings
4. Execute set DNS with dhcp set to false and updated ip address.
5. Verify Get DNS api to verify updated Ip address.
6.
Expected result: User should be able to enable/disable DHCP and updated ip address without any issues.
Actual result: On Geo vision camera it fails to set IP address, and on Bosch camera not able to enable/disable DHCP option.

attached set DNS api screenshot.
[image]

vyshali-chitikeshi, 2022-10-27:
Issue reproducible on Geo vision camera using latest images on 10/26/22.

ajcasagrande, 2022-10-27:
@vyshali-chitikeshi did you try rebooting the cameras after setting the new DNS? Also in any case it sounds like an issue with the cameras onvif implementations and nothing the device service would have control over. If anything, this sounds more like errata that belongs in onvif-footnotes than issues.

surajitx-pal, 2022-10-27:
Levski Release : Bosch
Same issue : Enabling/disabling of DHCP not working

vyshali-chitikeshi, 2023-06-12:
Issue reproducible with V3 code. On Bosch camera not able to enable and disable DHCP with set DNS API
[image]