edgexfoundry/device-onvif-camera #26: Bosch camera- Set network interfaces API not working correctly

vyshali-chitikeshi, 2022-04-04:
Summary of the problem:
Not able to enable/disable DHCP with set network interfaces api on Bosch camera

Issue reproducible on below camera models:
Bosch DINION 6000HD

Repro steps:
1. Execute get network interfaces command to retrieve existing network interfaces configuration
2.  Execute set network interface to enable and/or dhcp option.
3. Execute get network interfaces command to verify updated network interfaces configuration
Expected result: User should be able to enable/disable DHCP on network interface using set network interfaces api.
Actual result:  Enabling/Disabling of DHCP fails.

attached set network interfaces api screenshot
[image]

surajitx-pal, 2022-10-27:
Levski Release : still same issue (Oct-26-2022, Images)

vyshali-chitikeshi, 2022-10-28:
Issue reproducible on Geo vision camera as well with images on 10/27/22.

vyshali-chitikeshi, 2023-01-11:
With Hikvision-DS-2DE2A404IW-DE3 camera, set network interfaces gives 200 to enable/disable DHCP. However after updating interface camera is not reachable and needs to reconnect. after reconnect, updated settings does not exist.

vyshali-chitikeshi, 2023-06-07:
Issue reproducible with V3 code