edgexfoundry/device-onvif-camera #27: Geo vision/Bosch camera- Set system date and time not working correctly

vyshali-chitikeshi, 2022-04-04:
Summary of the problem:
Set system date and time API not working as expected .

Issue reproducible on below camera models:
Geo Vision GV-BX8700
On Geo Vision camera, NTP enable/disable works correctly but not able to set date manually.

Bosch DINION 6000HD
On Bosch camera, NTP enable/disable does not work and not able to set date manually.

Repro steps:

1. Execute get system date and time api to retrieve current camera date and time settings.
2. Enable/disable NTP and try updating system date and time using set system date and time API.
3. Execute get system date and time to verify updated date and time settigns.

Expected result: User able to set date NTP settings and set date manually.
Actual result: 
On Geo Vision camera, NTP enable/disable works correctly but not able to set date manually.
On Bosch camera, NTP enable/disable does not work and not able to set date manually.

attached set system date and time APi screenshot
[image]
.

vyshali-chitikeshi, 2022-10-27:
Issue reproducible on Geo vision camera using image on 10/26/22

surajitx-pal, 2022-10-27:
Levski Release - Bosch DINION 6000HD
1. NTP enable/disable does not work.
 2. Now able to set date manually, but **DaylightSavings** setting always false.

vyshali-chitikeshi, 2023-06-07:
Issue reproducible with V3 code on Geo vision camera