edgexfoundry/device-onvif-camera #28: Geo vision/Honey well/Bosch camera- Delete user not working

vyshali-chitikeshi, 2022-04-04:
Summary of the problem:
'Delete Users' API not deleting users.

Issue reproducible on below camera models:
Geo Vision GV-BX8700
Bosch DINION 6000HD
Honeywell HC30WB5R1

Repro steps:
1) add new test user as user/orator 'add user' api
2) Execute get users to verify newly created user details
3) Delete test user which is created in step 1 using 'Delete Users' API.
4) Execute get users api to verify deletion of test user.

Expected result: Delete users api deletes user successfully
Actual result: Not getting any error after executing delete users api, but its not deleting user.

Attached delete user API screenshot.
[image]

weichou1229, 2022-04-15:
According to the spec https://www.onvif.org/ver10/device/wsdl/devicemgmt.wsdl#op.DeleteUsers
```
[DeleteUsers]
      Username - unbounded; [string]
```
We put the following JSON data to delete users.
```
curl --location --request PUT 'http://0.0.0.0:59882/api/v2/device/name/Camera003/DeleteUsers' \
--header 'Content-Type: application/json' \
--data-raw '{
    "DeleteUsers": {
        "Username": ["Test1", "Test2"]
    }
}'
```

vyshali-chitikeshi, 2022-04-21:
Verified on Geo vision and it works correctly. will test on Honeywell and Bosch cameras

weichou1229, 2022-04-21:
I will add the postman script at this issue https://github.com/edgexfoundry-holding/device-onvif-camera/issues/47, I need time to verify the APIs with other issues.

brianointel, 2022-04-22:
verified in postman the following works for the boch camera

{
    "DeleteUsers": {
        "Username": ["Test1", "Test2"]
    }
}

vyshali-chitikeshi, 2022-04-25:
This issue can be closed

closed 2022-04-26 by weichou1229