edgexfoundry/device-onvif-camera #29: Geo vision/Honey well/Bosch camera- Set video encoder configuration not working correctly

vyshali-chitikeshi, 2022-04-04:
Summary of the problem:
Not able to update video encoder configuration using 'set video encoder configuration' API

Issue reproducible on below camera models:
Geo Vision GV-BX8700
Bosch DINION 6000HD
Honeywell HC30WB5R1

Repro steps:
1. Execute Get video encoder configuration API to retrieve existing video encoder configuration details
2.  Update video encoder configuration using 'set video encoder configuration'. For e.g. Govlength different than default
4. Intermittently its giving SOAP error on some cameras (Bosch), sometimes it wont give any error but not updating video encoder values as expected.  

Expected result: able to update video encoder configuration values correctly
Actual result: set video encoder configuration api not working as expected.

surajitx-pal, 2022-10-27:
Levski Release - Bosch,  Oct-26-2022 code freeze at main branch - same issue.

vyshali-chitikeshi, 2022-10-28:
Issue reproducible with images on 10/27/22 on Geovision camera.

vyshali-chitikeshi, 2023-01-12:
on Hikvision-DS-2DE2A404IW-DE3 camera, it gives invalid arg value error even tried to update with valid parameter value.
attached is the api output.
[image]

vyshali-chitikeshi, 2023-06-07:
Issue reproducible with V3 code