edgexfoundry/device-onvif-camera #32: Bosch camera- Set metadata configuration now working - soap error

surajitx-pal, 2022-04-05:
Summary of the problem:
'Set metadata configuration' API not working.

Issue reproducible on below camera models:
#Bosch DINION 6000HD

Reproduce steps:
Executed this below command to set-meta data configuration -

curl -C PUT http://localhost:59882/api/v2/device/name/Bosch-DINION-IP-starlight-6000-HD-00075fd8-7c31-317c-d85f-0700075fd85f/MetadataConfiguration \
    --header 'Content-Type: application/json' \
    --data-raw '
   {
    "MetadataConfiguration": {
        "Configuration": {
            "Token": "1",
            "SessionTimeout": "PT60S"
        }
    }
}'

Expected result: set users api will set metadata configuration successfully
Actual result:  - fault code: SOAP-ENV:Receiver ter:Action -500 return code.

surajitx-pal, 2022-10-27:
Levski Release (Oct-26-2022) - Same issue.

vyshali-chitikeshi, 2022-10-28:
On Geo vision camera: Using images on 10/27/22, set metadata configuration gives status code 200 when tried to change configuration->Multicast->Auto start to true. But get meta data configuration still shows old value which is configuration->multicast->auto start set to false. Same thing with changing session timeout value, setmetadata configuration api gives 200 but its not actually updating session timeout value.

vyshali-chitikeshi, 2023-01-12:
On Hikvision-DS-2DE2A404IW-DE3 camera, set metadata configuraiton fails with error code 500 even with valid parameter values. Below is the error
{
    "apiVersion": "v2",
    "message":"request failed, status code: 500, err: {\"apiVersion\":\"v2\",\"message\":\"error writing DeviceResourece MetadataConfiguration for HIKVISION-DS-2DE2A404IW-DE3-099f4000-4d50-11b4-82c8-c06ded544d95 -\> failed to execute write command, \<nil\> -\> failed to execute the request for the function 'SetMetadataConfiguration' of web service 'Media'. Onvif error: fault reason: the parameter value is illegal, fault detail: , fault code: env:Sender ter:InvalidArgVal ter:InvalidParameter\",\"statusCode\":500}",
    "statusCode": 500
}

lenny-intel, 2023-01-25:
Update Matrix