edgexfoundry/device-onvif-camera #33: Bosch camera- Configuration Analytics profile-Remove Configuration not removeing newly added configuration.

surajitx-pal, 2022-04-05:
Summary of the problem:
'Remove Configuration' API not working.

Issue reproducible on below camera models:
#Bosch DINION 6000HD

Reproduce steps:

   Execute this command to set Set metadata configuration

  command -add configuration -
 
 Step #1:
  
 curl -X PUT http://localhost:59882/api/v2/device/name/Bosch-DINION-IP-starlight-6000-HD-00075fd8-7c31-317c-d85f-0700075fd85f/AddConfiguration \
    --header 'Content-Type: application/json' \
    --data-raw  '
    {
   "AddConfiguration":{
      "ProfileToken":"0",
      "Name":"Analytics #3",
      "Configurations":[
         {
            "Type":"tt:Viproc",
            "Token":"1"
         }
      ]
   }
}'

Step #2
Verify with getProfiles command - new configuraiton added sccess fully.
curl -X GET http://localhost:59882/api/v2/device/name/Bosch-DINION-IP-starlight-6000-HD-00075fd8-7c31-317c-d85f-0700075fd85f/Media2Profiles

Step #3
Command to remove the newly added configuration -

curl -X PUT http://localhost:59882/api/v2/device/name/Bosch-DINION-IP-starlight-6000-HD-00075fd8-7c31-317c-d85f-0700075fd85f/RemoveConfiguration\
    --header 'Content-Type: application/json' \
    --data-raw  '
    {
    "RemoveConfiguration": {
        "ProfileToken": "0",
        "Configurations": [
            {
                "Type": "tt:Viproc",
                "Token": "1"
            }
        ]
    }
}'

Executed successfully - but not removed properly verify with GetProfiles command-

vyshali-chitikeshi, 2023-01-13:
On Hikvision-DS-2DE2A404IW-DE3 camera also, it gives status code 200 with remove configuration but its not actually removing configuration from media profile.

vyshali-chitikeshi, 2023-06-12:
Issue still exists with V3 code