edgexfoundry/device-onvif-camera #34: Bosch camera- Modify Analytics Modules not working.

surajitx-pal, 2022-04-05:
Summary of the problem:
'Modify Analytics Modules' API not working.

Issue reproducible on below camera models:
#Bosch DINION 6000HD

Reproduce steps:

   Execute this command to Modify Analytics Modules
   
   
curl -X PUT http://localhost:59882/api/v2/device/name/Bosch-DINION-IP-starlight-6000-HD-00075fd8-7c31-317c-d85f-0700075fd85f/AnalyticsModules \
    --header 'Content-Type: application/json' \
    --data-raw '
    
    {
    "AnalyticsModules": {
        "ConfigurationToken": "1",
        "AnalyticsModule": [
            {
                "Name": "Viproc",
                "Parameters": {
                    "SimpleItem": [
                        {
                            "Name": "Mode",
                            "Value": "Profile 144"
                        }
                    ]
                }
            }
        ]
    }
}'

 
Actual result - Soap error, 500
Executed - should be successfully modify

surajitx-pal, 2022-10-27:
Levski Release (Oct-26-2022 Image/main branch) - same issue reproduceable.

vyshali-chitikeshi, 2023-06-12:
Issue reproducible with V3 code. Find attached
[image]