edgexfoundry/device-onvif-camera #37: Bosch camera- modify Rule - If multiple same rule added then no modfication but if only one time added a rule then it works.

surajitx-pal, 2022-04-05:
Summary of the problem:
'modify Rule' API not working for same rule if added multiple times.

Issue reproducible on below camera models:
#Bosch DINION 6000HD

Reproduce steps:
   Execute this command 
  
  Step #1 
   created rule multiple times with same parameter- same rule are added as previous issue report.(issue #35)
   
   Step #2
   Modifying the same added rule -
   
   
curl -X PUT http://localhost:59882/api/v2/device/name/Bosch-DINION-IP-starlight-6000-HD-00075fd8-7c31-317c-d85f-0700075fd85f/Rules \
    --header 'Content-Type: application/json' \
    --data-raw '
    {
    "Rules": {
        "ConfigurationToken": "1",
        "Rule": [
            {
                "Name": "Test1",
                "Type": "tt:ObjectInField",
                "Parameters": {
                    "SimpleItem": [
                        {
                            "Name": "Armed",
                            "Value": "true"
                        }
                    ]
                }
            }
        ]
    }
}
    '

Then verify with GetRules command-
curl -x GET GET http://localhost:59882/api/v2/device/name/Bosch-DINION-IP-starlight-6000-HD-00075fd8-7c31-317c-d85f-0700075fd85f/Rules
 
Actual result - Not modifying rule, but working if rule is added one time only.
Executed - Should be modifying all same rule added before.

vyshali-chitikeshi, 2023-06-12:
Issue still exists with V3 code

cloudxxx8, 2023-06-19:
@vyshali-chitikeshi for this kind of issue, did you try to execute the API of camera directly?
If the camera doesn't follow the spec to implement the API, there is nothing Device Service can do.