edgexfoundry/device-onvif-camera #56: Core data client issues fatal errors

ajcasagrande, 2022-05-05:
What is this client used for? I have been running this service in an environment without core-data and the software stack works just fine. 

https://github.com/edgexfoundry-holding/device-onvif-camera/blob/72c31d4250f40c982c4b3924e6f71b4c9e8a6948/cmd/res/configuration.toml#L61-L64

However, something about the last 4 commits has broken it, as i get an endless stream of errors:

```
level=WARN ts=2022-05-05T03:47:31.959858833Z app=device-onvif-camera source=clients.go:137 msg="unable to Get service endpoint for 'core-data': no matching service endpoint found. retrying..."
level=WARN ts=2022-05-05T03:47:32.961778923Z app=device-onvif-camera source=clients.go:137 msg="unable to Get service endpoint for 'core-data': no matching service endpoint found. retrying..."
level=WARN ts=2022-05-05T03:47:33.962877041Z app=device-onvif-camera source=clients.go:137 msg="unable to Get service endpoint for 'core-data': no matching service endpoint found. retrying..."
level=WARN ts=2022-05-05T03:47:34.964178535Z app=device-onvif-camera source=clients.go:137 msg="unable to Get service endpoint for 'core-data': no matching service endpoint found. retrying..."
level=WARN ts=2022-05-05T03:47:35.965569188Z app=device-onvif-camera source=clients.go:137 msg="unable to Get service endpoint for 'core-data': no matching service endpoint found. retrying..."
level=WARN ts=2022-05-05T03:47:36.966632657Z app=device-onvif-camera source=clients.go:137 msg="unable to Get service endpoint for 'core-data': no matching service endpoint found. retrying..."
level=WARN ts=2022-05-05T03:47:37.967566331Z app=device-onvif-camera source=clients.go:137 msg="unable to Get service endpoint for 'core-data': no matching service endpoint found. retrying..."
level=WARN ts=2022-05-05T03:47:38.968655964Z app=device-onvif-camera source=clients.go:137 msg="unable to Get service endpoint for 'core-data': no matching service endpoint found. retrying..."
level=WARN ts=2022-05-05T03:47:39.969798371Z app=device-onvif-camera source=clients.go:137 msg="unable to Get service endpoint for 'core-data': no matching service endpoint found. retrying..."
```

I think it may be related to the security hardening PR. The service no longer starts.

weichou1229, 2022-05-06:
@ajcasagrande 
1. The core-data config is required for device-sdk-go https://github.com/edgexfoundry/device-sdk-go/blob/main/internal/clients/init.go#L58

2. And we just upgrade the lib from v2.1 to v2.2, it's the latest changes from the go-mod-bootstrap, the BootstrapHandler keep trying to get the service endpoint.
https://github.com/edgexfoundry/go-mod-bootstrap/pull/305/files

weichou1229, 2022-05-06:
@ajcasagrande 
I opened the issue on go-mod-bootstrap.
https://github.com/edgexfoundry/go-mod-bootstrap/issues/335

lenny-intel, 2022-05-09:
If Core data is not being used, then it needs to be removed from the Clients section of the configuration.

Per the Registry ADR, bootstrap now uses the registry to get the endpoint of all clients.

lenny-intel, 2022-05-09:
> 1. The core-data config is required for device-sdk-go https://github.com/edgexfoundry/device-sdk-go/blob/main/internal/clients/init.go#L58

@weichou1229,  Since Core Data is now optional, this check in the SDK should be removed.

PR #61 (2022-05-10): build: Upgrade sdk to v2.2.0-dev.38
Close #56

## PR Checklist
Please check if your PR fulfills the following requirements:

- [ ] Tests for the changes have been added (for bug fixes / features) **not impact the test**
- [ ] Docs have been added / updated (for bug fixes / features) **not impact the doc**

**If your build fails** due to your commit message not passing the build checks, please review the guidelines here: https://github.com/edgexfoundry/edgex-examples/blob/master/.github/CONTRIBUTING.md.

## What is the current behavior?

## Issue Number: #56

## What is the new behavior?
Upgrad [... cut]
commit 5006a0cf (2022-05-10): build: Upgrade sdk to v2.2.0-dev.38

Close #56
closed 2022-05-10 by cloudxxx8