edgexfoundry/device-onvif-camera #69: Remove Docker Compose folder now that this is in Compose Builder

lenny-intel, 2022-05-23:
# 🚀 Feature Request

### Relevant Package [**REQUIRED**]
doc/docker-compose folder

### Description [**REQUIRED**]
This is no longer needed now that Compose Builder in edgex-compose has been update to included this

### Describe the solution you'd like
remove folder and update main README to describe how to use Compose Builder, including any needed Env Overrides or changes in Consul

### Describe alternatives you've considered
 Have you considered any alternative solutions or workarounds?

lenny-intel, 2022-06-23:
@cloudxxx8 , @weichou1229 The Intel team has prioritized the remaining issues. 
Looking to fix all marked as `High` prior to the first release. 
The team would like your help with this one and will handle the remaining Highs. THX!

weichou1229, 2022-06-24:
No problem, will do.

commit 82843687 (2022-07-06): docs: Remove compose folder and update doc (#95)

Close #69