edgexfoundry/device-onvif-camera #73: Error in Auto Discovery of Camera

tapeshmathur, 2022-05-30:
🛑
I followed : https://github.com/edgexfoundry/device-onvif-camera/blob/main/doc/auto-discovery-docker.md ( i.e. Set Up Auto Discovery with Docker)
When I am running the containers I am getting a error in device-onvif-camera:
level=ERROR ts=2022-05-30T04:13:44.682604091Z app=device-onvif-camera source=init.go:42 msg="Failed to init cache: failed to send a http request -> Get \"http://edgex-core-metadata:59881/api/v2/device/service/name/device-onvif-camera?limit=-1&offset=0\": dial tcp: lookup edgex-core-metadata on 192.168.0.1:53: read udp 192.168.0.115:33190->192.168.0.1:53: i/o timeout"
level=INFO ts=2022-05-30T04:13:44.682709107Z app=device-onvif-camera source=bootstrap.go:135 msg="Un-Registering service from the Registry"

AND

I followed: https://github.com/edgexfoundry/device-onvif-camera/blob/main/doc/auto-discovery.md (i.e. Auto Discovery)
When I am running the containers I am getting a error in device-onvif-camera:
level=INFO ts=2022-05-30T04:31:51.942762705Z app=device-onvif-camera source=devices.go:49 msg="Loading pre-defined devices from /res/devices"
level=INFO ts=2022-05-30T04:31:51.943721555Z app=device-onvif-camera source=devices.go:85 msg="Device device-onvif-camera exists, using the existing one"
level=INFO ts=2022-05-30T04:31:51.943814738Z app=device-onvif-camera source=message.go:50 msg="Service dependencies resolved..."
level=INFO ts=2022-05-30T04:31:51.943840422Z app=device-onvif-camera source=message.go:51 msg="Starting device-onvif-camera 0.0.0 "
level=INFO ts=2022-05-30T04:31:51.944104305Z app=device-onvif-camera source=message.go:55 msg="Camera device service started"
level=INFO ts=2022-05-30T04:31:51.944130546Z app=device-onvif-camera source=message.go:58 msg="Service started in: 1.279078281s"
level=INFO ts=2022-05-30T04:31:51.94394255Z app=device-onvif-camera source=autodiscovery.go:51 msg="Starting auto-discovery with duration 10s"
level=DEBUG ts=2022-05-30T04:31:51.944162915Z app=device-onvif-camera source=discovery.go:34 msg="protocol discovery triggered"
route ip+net: no such network interface
level=DEBUG ts=2022-05-30T04:31:52.946479407Z app=device-onvif-camera source=async.go:127 msg="Filtered device addition finished"
level=DEBUG ts=2022-05-30T04:32:02.946596983Z app=device-onvif-camera source=discovery.go:34 msg="protocol discovery triggered"
route ip+net: no such network interface
level=DEBUG ts=2022-05-30T04:32:03.949196507Z app=device-onvif-camera source=async.go:127 msg="Filtered device addition finished"
level=DEBUG ts=2022-05-30T04:32:13.949314222Z app=device-onvif-camera source=discovery.go:34 msg="protocol discovery triggered"
route ip+net: no such network interface

Can someone please help me in resolving the above mentioned errors or what can be the cause of the issue?
Thanks in advance.

cloudxxx8, 2022-05-30:
are you running device-onvif-camera and other services in the same docker network?

cloudxxx8, 2022-05-30:
please verify whether you can access the core-metadata API in another way, such as browser, curl, or postman
http://edgex-core-metadata:59881/api/v2/device/service/name/device-onvif-camera?limit=-1&offset=0

tapeshmathur, 2022-05-31:
> please verify whether you can access the core-metadata API in another way, such as browser, curl, or postman http://edgex-core-metadata:59881/api/v2/device/service/name/device-onvif-camera?limit=-1&offset=0

When I do a curl request to  http://edgex-core-metadata:59881/api/v2/device/service/name/device-onvif-camera?limit=-1&offset=0 , it is giving an error as cannot resolve edgex-core-metadata. But if do curl on http://<Machine IP>:59881/api/v2/device/service/name/device-onvif-camera?limit=-1&offset=0 , I am getting a response.

cloudxxx8, 2022-06-02:
you can just use the docker port mapping, and the host ip should work
in the configuration file https://github.com/edgexfoundry/device-onvif-camera/blob/18118626ca0893b728531bfc2ef8f97357a298a2/cmd/res/configuration.toml#L25
or, you can set the environment variable in the docker container to override the config SERVICE_HOST
like https://github.com/edgexfoundry/device-onvif-camera/blob/18118626ca0893b728531bfc2ef8f97357a298a2/doc/docker-compose/add-device-onvif-camera.yml#L34

weichou1229, 2022-08-23:
@tapeshmathur device-onvif-camera has a new discovery mechanism `netscan`, it can cross different networks, 
```
APPCUSTOM_DISCOVERYSUBNETS: 192.168.93.0/24
```
see https://github.com/edgexfoundry/device-onvif-camera/blob/main/doc/auto-discovery.md#netscan

lenny-intel, 2023-01-25:
Closing due to no response

closed 2023-01-25 by lenny-intel