edgexfoundry/device-onvif-camera #96: netscan discovery was called, but DiscoverySubnets are empty!

wangyongxiao, 2022-07-01:
# 📚 Docs or Wiki Bug Report

### Description [**REQUIRED**]

 According to the steps of  
[auto-discovery-docker.md](https://github.com/edgexfoundry/device-onvif-camera/blob/main/doc/auto-discovery-docker.md)  i get an msg below:

`level=WARN ts=2022-07-01T07:46:12.894347595Z app=device-onvif-camera source=driver.go:580 msg="netscan discovery was called, but DiscoverySubnets are empty!"`
 
and i found the DiscoverySubnets is empty  in configration.toml which is in device-onvif-camera container

and i can not change it because it's read_only

what can i do now?

weichou1229, 2022-07-01:
will update the doc via this PR https://github.com/edgexfoundry/device-onvif-camera/pull/95

closed 2022-07-01 by wangyongxiao

wangyongxiao, 2022-07-01:
@weichou1229  can you tell me where  to find APPCUSTOM_DISCOVERYSUBNETS doc?
i can't find it anywhere i konw

weichou1229, 2022-07-03:
See https://github.com/edgexfoundry/device-onvif-camera/blob/main/cmd/res/configuration.toml#L120