edgexfoundry/device-onvif-camera #132: GetSnapshot命令执行报错

447264188, 2022-08-12:
## GetSnapshot命令执行报错
测试人：thundersoft
测试环境：EdgeX2.2 
摄像头品牌型号：HIKVISION-DS-2DC4223IW-DE，HIKVISION-DS-2CD3T27FWDA4-LS
测试过程：
海康摄像头在执行 GetSnapshot 时，如果 "mode" 设置为 both 和 digest 会报错，设置为usernametoken 则可以正常执行。
另外，做了一下对比测试。使用ONVIF Device Test Tool 中进行测试则 usernametoken  和 digest 都可正常执行。
配置信息如下：

```toml
    [Writable.InsecureSecrets.Camera001]
    path = "default"
      [Writable.InsecureSecrets.Camera001.Secrets]
      username = "admin"
      password = "123456"
      mode = "both" # assign "digest" | "usernametoken" | "both" | "none"
```
测试脚本如下：
```bash
curl --location --request GET 'http://10.0.20.42:59882/api/v2/device/name/HIKVISION-DS-2DC4223IW-DE-3bd4c000-b1b9-11b3-8202-4cf5dc64a22c/Snapshot'
```
错误信息如下：

```json
{
    "apiVersion": "v2",
    "message":"request failed, status code: 500, err: {\"apiVersion\":\"v2\",\"message\":\"error reading DeviceResourece Snapshot for HIKVISION-DS-2DC4223IW-DE-3bd4c000-b1b9-11b3-8202-4cf5dc64a22c -\> failed to execute read command -\> http request for image failed with status 401, \<!DOCTYPE html\>\\r\\n\<html\>\<head\>\<title\>Document Error: Unauthorized\</title\>\</head\>\\r\\n\<body\>\<h2\>Access Error: 401 -- Unauthorized\</h2\>\\r\\n\<p\>Authentication Error: Digest authentication information verification failed\</p\>\\r\\n\</body\>\\r\\n\</html\>\\r\\n\",\"statusCode\":500}",
    "statusCode": 500
}
```

cloudxxx8, 2022-08-14:
Please attach the debug log of this device service for all your scenarios.
Also, please use English to report the issue, the engineers around the world can provide the help.

lenny-intel, 2023-01-25:
closing for no activity

closed 2023-01-25 by lenny-intel