edgexfoundry/device-onvif-camera #134: When set discovery mode to 'netscan', not get any devices,Please help!

edgego, 2022-08-15:
Dear :
  Run the  service ( version 2.2.0 ) via  docker-compose , did not  discovery any devices , but indeed there is a hikvsion ip camera with ip 192.168.1.64. Following is the configuration for AppCustom
     [AppCustom]
       DiscoverySubnets = "192.168.1.0/24"
       ProbeAsyncLimit = 4000
       CredentialsRetryTime = 120
       CredentialsRetryWait = 1
       RequestTimeout = 5
       DiscoveryEthernetInterface = "eth0"
       DefaultSecretPath = "credentials001"
       BaseNotificationURL = "http://support-notifications:59860"
       DiscoveryMode = "netscan"
       ProbeTimeoutMillis = 2000
       EnableStatusCheck = true
       CheckStatusInterval = 30
       MaxDiscoverDurationSeconds = 300

Following is the debug information:
level=INFO ts=2022-08-15T12:10:22.184536316Z app=device-onvif-camera source=message.go:55 msg="device-onvif-camera started"
level=INFO ts=2022-08-15T12:10:22.18454203Z app=device-onvif-camera source=message.go:58 msg="Service started in: 14.103422ms"
level=INFO ts=2022-08-15T12:10:22.184541862Z app=device-onvif-camera source=autodiscovery.go:51 msg="Starting auto-discovery with duration 1h0m0s"
level=DEBUG ts=2022-08-15T12:10:22.18458158Z app=device-onvif-camera source=discovery.go:34 msg="protocol discovery triggered"
level=INFO ts=2022-08-15T12:10:22.184595131Z app=device-onvif-camera source=driver.go:536 msg="Discover was called."
level=INFO ts=2022-08-15T12:10:22.184606648Z app=device-onvif-camera source=driver.go:259 msg="Adding provision watchers from res/provision_watchers"
level=DEBUG ts=2022-08-15T12:10:22.184787006Z app=device-onvif-camera source=driver.go:273 msg="processing res/provision_watchers/device-onvif-camera-test-watcher.json"
level=DEBUG ts=2022-08-15T12:10:22.184954671Z app=device-onvif-camera source=driver.go:293 msg="skip existing provision watcher test"
level=DEBUG ts=2022-08-15T12:10:22.185024823Z app=device-onvif-camera source=discover.go:31 msg="AutoDiscover called with the following parameters: {Subnets:[192.168.1.0/24] ScanPorts:[3702] AsyncLimit:4000 NetworkProtocol:udp Timeout:2s Logger:{owningServiceName:device-onvif-camera logLevel:0xc0004750f0 rootLogger:0xc000408c30 levelLoggers:map[DEBUG:0xc000408cd0 ERROR:0xc000408dc0 INFO:0xc000408d20 TRACE:0xc000408c80 WARN:0xc000408d70]}}"
level=DEBUG ts=2022-08-15T12:10:22.185038949Z app=device-onvif-camera source=discover.go:87 msg="total estimated network probes: 254, async limit: 254, probe timeout: 2s, estimated time: 2s"
level=TRACE ts=2022-08-15T12:10:22.185359418Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.1:3702"
level=TRACE ts=2022-08-15T12:10:22.185739406Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.2:3702"
level=TRACE ts=2022-08-15T12:10:22.18619483Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.3:3702"
level=TRACE ts=2022-08-15T12:10:22.186642311Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.4:3702"
level=TRACE ts=2022-08-15T12:10:22.187018753Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.5:3702"
level=TRACE ts=2022-08-15T12:10:22.187397892Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.6:3702"
level=TRACE ts=2022-08-15T12:10:22.187592766Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.7:3702"
level=TRACE ts=2022-08-15T12:10:22.187628604Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.8:3702"
level=TRACE ts=2022-08-15T12:10:22.187897443Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.9:3702"
level=TRACE ts=2022-08-15T12:10:22.187904659Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.10:3702"
level=TRACE ts=2022-08-15T12:10:22.188047257Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.11:3702"
level=TRACE ts=2022-08-15T12:10:22.188141461Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.12:3702"
level=TRACE ts=2022-08-15T12:10:22.188850496Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.15:3702"
level=TRACE ts=2022-08-15T12:10:22.189644804Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.16:3702"
level=TRACE ts=2022-08-15T12:10:22.189773227Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.17:3702"
level=TRACE ts=2022-08-15T12:10:22.189841977Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.18:3702"
level=TRACE ts=2022-08-15T12:10:22.189865261Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.19:3702"
level=TRACE ts=2022-08-15T12:10:22.189884552Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.20:3702"
level=TRACE ts=2022-08-15T12:10:22.1902141Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.21:3702"
level=TRACE ts=2022-08-15T12:10:22.190234829Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.22:3702"
level=TRACE ts=2022-08-15T12:10:22.190322402Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.23:3702"
level=TRACE ts=2022-08-15T12:10:22.190344369Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.24:3702"
level=TRACE ts=2022-08-15T12:10:22.190429347Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.25:3702"
level=TRACE ts=2022-08-15T12:10:22.190656388Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.26:3702"
level=TRACE ts=2022-08-15T12:10:22.1907741Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.27:3702"
level=TRACE ts=2022-08-15T12:10:22.190794198Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.28:3702"
level=TRACE ts=2022-08-15T12:10:22.190833186Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.29:3702"
level=TRACE ts=2022-08-15T12:10:22.190847775Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.30:3702"
level=TRACE ts=2022-08-15T12:10:22.189602805Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.13:3702"
level=TRACE ts=2022-08-15T12:10:22.190885849Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.31:3702"
level=TRACE ts=2022-08-15T12:10:22.188667378Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.14:3702"
level=TRACE ts=2022-08-15T12:10:22.193837141Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.142:3702"
level=TRACE ts=2022-08-15T12:10:22.193928146Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.210:3702"
level=TRACE ts=2022-08-15T12:10:22.19394977Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.199:3702"
level=TRACE ts=2022-08-15T12:10:22.194081357Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.205:3702"
level=TRACE ts=2022-08-15T12:10:22.194198226Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.35:3702"
level=TRACE ts=2022-08-15T12:10:22.194281742Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.149:3702"
level=TRACE ts=2022-08-15T12:10:22.194319403Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.217:3702"
level=TRACE ts=2022-08-15T12:10:22.194445676Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.102:3702"
level=TRACE ts=2022-08-15T12:10:22.194457048Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.32:3702"
level=TRACE ts=2022-08-15T12:10:22.194668257Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.216:3702"
level=TRACE ts=2022-08-15T12:10:22.194720779Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.103:3702"
[... 25 lines cut ...]
level=TRACE ts=2022-08-15T12:10:22.279490166Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.245:3702"
level=TRACE ts=2022-08-15T12:10:22.279722319Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.247:3702"
level=TRACE ts=2022-08-15T12:10:22.279989928Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.248:3702"
level=TRACE ts=2022-08-15T12:10:22.280045717Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.250:3702"
level=TRACE ts=2022-08-15T12:10:22.280316526Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.189:3702"
level=TRACE ts=2022-08-15T12:10:22.280585062Z app=device-onvif-camera source=discover.go:192 msg="Dial: 192.168.1.221:3702"
[... cut ...]
level=TRACE ts=2022-08-15T12:10:24.281141973Z app=device-onvif-camera source=onvifdiscovery.go:209 msg="192.168.1.252:3702: No Response"
lmsg="192.168.1.112:3702: No Response"
level=DEBUG ts=2022-08-15T12:10:24.301456498Z app=device-onvif-camera source=driver.go:633 msg="NetScan result: []"
level=INFO ts=2022-08-15T12:10:24.301466184Z app=device-onvif-camera source=driver.go:634 msg="Discovered 0 device(s) in 2.116462574s via netscan."
level=DEBUG ts=2022-08-15T12:10:24.301484293Z app=device-onvif-camera source=async.go:127 msg="Filtered device addition finished"

ajcasagrande, 2022-08-15:
**I believe this is related to another auto-discovery issue I am tracking down. If that is the case, it should be fixed in the next few days.**

Here is something to try in the meantime to check if it will work with the work-in-progress discovery fix:

Open a UDP connection to the camera on port 3702 using netcat:
```shell
nc -u 192.168.1.64 3702
```

Paste this entire xml message as-is, and then hit Enter if needed.
```xml
<?xml version="1.0" encoding="UTF-8"?>
<soap-env:Envelope xmlns:soap-env="http://www.w3.org/2003/05/soap-envelope" xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:soap-enc="http://www.w3.org/2003/05/soap-encoding">
   <soap-env:Header>
      <a:Action mustUnderstand="1">http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
      <a:MessageID>uuid:a277f13a-ecae-4492-9d6a-218982122d1c</a:MessageID>
      <a:To mustUnderstand="1">urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
   </soap-env:Header>
   <soap-env:Body>
      <d:Probe>
         <d:Types>dn:NetworkVideoTransmitter</d:Types>
      </d:Probe>
   </soap-env:Body>
</soap-env:Envelope>
```

You should hopefully get an xml message back from the camera, containing things such as `ProbeMatch` and `XAddrs` if successful. If you wouldn't mind posting that here, that would be helpful as well.

`Ctrl-C` to close the connection.

edgego, 2022-08-16:
@ajcasagrande  thanks for replay, following is the back :

<?xml version="1.0" encoding="UTF-8"?>
<env:Envelope xmlns:env="http://www.w3.org/2003/05/soap-envelope" xmlns:soapenc="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:timg="http://www.onvif.org/ver20/imaging/wsdl" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:tan="http://www.onvif.org/ver20/analytics/wsdl" xmlns:tst="http://www.onvif.org/ver10/storage/wsdl" xmlns:ter="http://www.onvif.org/ver10/error" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:tns1="http://www.onvif.org/ver10/topics" xmlns:tmd="http://www.onvif.org/ver10/deviceIO/wsdl" xmlns:wsdl="http://schemas.xmlsoap.org/wsdl" xmlns:wsoap12="http://schemas.xmlsoap.org/wsdl/soap12" xmlns:http="http://schemas.xmlsoap.org/wsdl/http" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:wsadis="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:wsa="http://www.w3.org/2005/08/addressing" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:wsrf-bf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:wsntw="http://docs.oasis-open.org/wsn/bw-2" xmlns:wsrf-rw="http://docs.oasis-open.org/wsrf/rw-2" xmlns:wsaw="http://www.w3.org/2006/05/addressing/wsdl" xmlns:wsrf-r="http://docs.oasis-open.org/wsrf/r-2" xmlns:trc="http://www.onvif.org/ver10/recording/wsdl" xmlns:tse="http://www.onvif.org/ver10/search/wsdl" xmlns:trp="http://www.onvif.org/ver10/replay/wsdl" xmlns:tnshik="http://www.hikvision.com/2011/event/topics" xmlns:hikwsd="http://www.onvifext.com/onvif/ext/ver10/wsdl" xmlns:hikxsd="http://www.onvifext.com/onvif/ext/ver10/schema" xmlns:tas="http://www.onvif.org/ver10/advancedsecurity/wsdl" xmlns:tr2="http://www.onvif.org/ver20/media/wsdl" xmlns:axt="http://www.onvif.org/ver20/analytics"><env:Header><wsadis:MessageID></wsadis:MessageID>
<wsadis:RelatesTo>uuid:4d44d0f8-677c-4bd7-87fb-5f57ed8bda75</wsadis:RelatesTo>
<wsadis:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/fault</wsadis:Action>
</env:Header>
<env:Body><env:Fault><env:Code><env:Value>env:Sender</env:Value>
<env:Subcode><env:Value>d:MatchingRuleNotSupported</env:Value>
</env:Subcode>
</env:Code>
<env:Reason><env:Text xml:lang="en">the xml format error</env:Text>
</env:Reason>
</env:Fault>
</env:Body>
</env:Envelope>

ajcasagrande, 2022-08-16:
@edgego That is an interesting response, I have not seen that before. The `MatchingRuleNotSupported` leads me to believe that it doesn't support filtering by device type (ie. `NetworkVideoTransmitter`), but the `<env:Text xml:lang="en">the xml format error</env:Text>` makes me think it did not like the xml value sent in.

Can you try sending the following different messages and seeing if any of them respond better? Make sure to close the connection and start a new one each time using the `nc -u 192.168.1.64 3702`

> NOTE: It is possible sometimes the camera will not respond anything back. Assume that means it didn't like the message.

without xml header:
```
<soap-env:Envelope xmlns:soap-env="http://www.w3.org/2003/05/soap-envelope" xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:soap-enc="http://www.w3.org/2003/05/soap-encoding">
   <soap-env:Header>
      <a:Action mustUnderstand="1">http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
      <a:MessageID>uuid:a277f13a-ecae-4492-9d6a-218982122d2c</a:MessageID>
      <a:To mustUnderstand="1">urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
   </soap-env:Header>
   <soap-env:Body>
      <d:Probe>
         <d:Types>dn:NetworkVideoTransmitter</d:Types>
      </d:Probe>
   </soap-env:Body>
</soap-env:Envelope>
```

without `dn:NetworkVideoTransmitter` filter
```
<?xml version="1.0" encoding="UTF-8"?>
<soap-env:Envelope xmlns:soap-env="http://www.w3.org/2003/05/soap-envelope" xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:soap-enc="http://www.w3.org/2003/05/soap-encoding">
   <soap-env:Header>
      <a:Action mustUnderstand="1">http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
      <a:MessageID>uuid:a277f13a-ecae-4492-9d6a-218982122d3c</a:MessageID>
      <a:To mustUnderstand="1">urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
   </soap-env:Header>
   <soap-env:Body>
      <d:Probe>
      </d:Probe>
   </soap-env:Body>
</soap-env:Envelope>
```

without xml header and without `dn:NetworkVideoTransmitter` filter
```
<soap-env:Envelope xmlns:soap-env="http://www.w3.org/2003/05/soap-envelope" xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:soap-enc="http://www.w3.org/2003/05/soap-encoding">
   <soap-env:Header>
      <a:Action mustUnderstand="1">http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
      <a:MessageID>uuid:a277f13a-ecae-4492-9d6a-218982122d4c</a:MessageID>
      <a:To mustUnderstand="1">urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
   </soap-env:Header>
   <soap-env:Body>
      <d:Probe>
      </d:Probe>
   </soap-env:Body>
</soap-env:Envelope>
```

edgego, 2022-08-17:
@ajcasagrande thanks , the last  without any new-lines packet works , response as following:
<?xml version="1.0" encoding="UTF-8"?>
<env:Envelope xmlns:env="http://www.w3.org/2003/05/soap-envelope" xmlns:soapenc="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:timg="http://www.onvif.org/ver20/imaging/wsdl" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:tan="http://www.onvif.org/ver20/analytics/wsdl" xmlns:tst="http://www.onvif.org/ver10/storage/wsdl" xmlns:ter="http://www.onvif.org/ver10/error" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:tns1="http://www.onvif.org/ver10/topics" xmlns:tmd="http://www.onvif.org/ver10/deviceIO/wsdl" xmlns:wsdl="http://schemas.xmlsoap.org/wsdl" xmlns:wsoap12="http://schemas.xmlsoap.org/wsdl/soap12" xmlns:http="http://schemas.xmlsoap.org/wsdl/http" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:wsadis="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:wsa="http://www.w3.org/2005/08/addressing" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:wsrf-bf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:wsntw="http://docs.oasis-open.org/wsn/bw-2" xmlns:wsrf-rw="http://docs.oasis-open.org/wsrf/rw-2" xmlns:wsaw="http://www.w3.org/2006/05/addressing/wsdl" xmlns:wsrf-r="http://docs.oasis-open.org/wsrf/r-2" xmlns:trc="http://www.onvif.org/ver10/recording/wsdl" xmlns:tse="http://www.onvif.org/ver10/search/wsdl" xmlns:trp="http://www.onvif.org/ver10/replay/wsdl" xmlns:tnshik="http://www.hikvision.com/2011/event/topics" xmlns:hikwsd="http://www.onvifext.com/onvif/ext/ver10/wsdl" xmlns:hikxsd="http://www.onvifext.com/onvif/ext/ver10/schema" xmlns:tas="http://www.onvif.org/ver10/advancedsecurity/wsdl" xmlns:tr2="http://www.onvif.org/ver20/media/wsdl" xmlns:axt="http://www.onvif.org/ver20/analytics"><env:Header><wsadis:MessageID>urn:uuid:617dc000-4574-11b4-82be-2428fdc5062f</wsadis:MessageID>
<wsadis:RelatesTo>uuid:a277f13a-ecae-4492-9d6a-218982122d2c</wsadis:RelatesTo>
<wsadis:To>http://schemas.xmlsoap.org/ws/2004/08/addressing/role/anonymous</wsadis:To>
<wsadis:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/ProbeMatches</wsadis:Action>
<d:AppSequence InstanceId="1660727855" MessageNumber="4"/>
</env:Header>
<env:Body><d:ProbeMatches><d:ProbeMatch><wsadis:EndpointReference><wsadis:Address>urn:uuid:617dc000-4574-11b4-82be-2428fdc5062f</wsadis:Address>
</wsadis:EndpointReference>
<d:Types>dn:NetworkVideoTransmitter tds:Device</d:Types>
<d:Scopes>onvif://www.onvif.org/type/video_encoder onvif://www.onvif.org/Profile/Streaming onvif://www.onvif.org/Profile/G onvif://www.onvif.org/Profile/T onvif://www.onvif.org/type/audio_encoder onvif://www.onvif.org/type/ptz onvif://www.onvif.org/MAC/24:28:fd:c5:06:2f onvif://www.onvif.org/hardware/DS-2DE2402IW-D3/W/XM onvif://www.onvif.org/name/HIKVISION%20DS-2DE2402IW-D3/W/XM onvif://www.onvif.org/location/city/hangzhou</d:Scopes>
<d:XAddrs>http://192.168.1.64/onvif/device_service</d:XAddrs>
<d:MetadataVersion>10</d:MetadataVersion>
</d:ProbeMatch>
</d:ProbeMatches>
</env:Body>
</env:Envelope>

edgego, 2022-08-17:
But inside docker container the command 'nc -u 192.168.1.64 3702' always hanging up after input soap packet. From outside dcoker container can get response

ajcasagrande, 2022-08-23:
@edgego please test with the latest docker images:

`nexus3.edgexfoundry.org:10004/device-onvif-camera:latest`
or
`nexus3.edgexfoundry.org:10004/device-onvif-camera:2.3.0-dev.8`

edgego, 2022-09-01:
@ajcasagrande  tested both images:nexus3.edgexfoundry.org:10004/device-onvif-camera:latest and nexus3.edgexfoundry.org:10004/device-onvif-camera:2.3.0-dev.8, still can not discovery camera . And also run command : nc -u 192.168.1.64 3702  inside the docker , hung up when input xml packet that you provided.
   Probably this issue is due to  WS-Discovery muiticast on 3702 only working on local docker network . But camera network same as host network, that is different network with docker network. So need implement remote discovery. 

Following two log files for both images.

[logs-from-device-onvif-camera-in-device-onvif-camera-8c567c86d-bd5k6.log](https://github.com/edgexfoundry/device-onvif-camera/files/9470060/logs-from-device-onvif-camera-in-device-onvif-camera-8c567c86d-bd5k6.log)
[logs-from-device-onvif-camera-in-device-onvif-camera-5b75dfd88c-vgmgc.log](https://github.com/edgexfoundry/device-onvif-camera/files/9470062/logs-from-device-onvif-camera-in-device-onvif-camera-5b75dfd88c-vgmgc.log)

ajcasagrande, 2022-09-02:
@edgego When using netscan there is no multicast message, but instead a unicast UDP probe over port 3702. This works behind the docker networks unlike the multicast version. There are many devices we have validated to work using this approach.

One potential cause of this is the camera is sending the response to the probe in a way or to an address that is not properly routable back to the docker container/device service. I have seen similar problems like this with the Happytime Simulator (it receives the probe, but the response is lost). This is why even when using `nc ...` you do not get a response in docker.

It might be possible to force it to respond properly by filling out the [`ReplyTo` fields of the probe](http://docs.oasis-open.org/ws-dd/discovery/1.1/wsdd-discovery-1.1-spec.html).

If we ever implement #137 that could help aleviate this issue.

edgego, 2022-09-03:
@ajcasagrande  You mean add a custome url  to field ReplyTo address? Then can get camera information from that custom url ?
Also change wsdiscovery.BuildProbeMessage  to  build soap probe request?

edgego, 2022-09-10:
@ajcasagrande  check them,probably can get some clue:

https://github.com/moby/moby/issues/15127
https://blog.birost.com/a?ID=01050-591e59cc-f901-4fde-ae3c-a64ea03a46ae

ajcasagrande, 2022-09-15:
thanks. A mixture of your links and playing around with the ReplyTo field, we may be able to resolve the issue. I do not own any cameras which exhibit this kind of behavior, but it may be possible to debug using the Happytime simulator.

@edgego Would you mind sharing the exact model number of your camera?

edgego, 2022-09-17:
@ajcasagrande  Thx, camera model  : hikvision  Ds-2DE2402Iw-D3/W/XM,seiral no : G29457470

edgego, 2022-09-20:
@ajcasagrande   Would you please tell how to get image resolution from dtos.Event reading ? Can get image from Snapshot comamnd , but no image resolution.  Thanks !

ajcasagrande, 2022-09-21:
@edgego In general, you should create separate Github Issues for these, so they can be tracked better.

The `GetSnapshot` command only sends the image binary. Currently due to #122 only the first profile is supported for getting the snapshot. So I believe you have 2 options:

1. Decode the snapshot and use a library that supports decoding image formats to read the image metadata, or
2. Call `GetProfiles` and find the video resolution of the first Profile. This should hopefully match the image resolution.

edgego, 2022-10-02:
Thanks , followed 1st option , resolved  image resolution issue

lenny-intel, 2022-10-03:
question answered, closing issue

closed 2022-10-03 by lenny-intel

ShubhamKPBoss, 2024-01-06:
Hello, can you please identify Where does the uuid come from? <a:MessageID>uuid:a277f13a-ecae-4492-9d6a-218982122d4c</a:MessageID>

lindseysimple, 2024-01-16:
@ShubhamKPBoss This issue has been closed and is not active. If you have any question for the issue assignee or reporter, please send an email to them directly, thanks.