edgexfoundry/device-onvif-camera #138: 'GetSnapshot' API can not working properly

447264188, 2022-08-18:
### Summary of the problem:
'GetSnapshot' API can not working properly

###  Issue reproducible on below camera models:
 Dahua-DH-P20A1

### Tester
from ThunderSoft

### Reproduce steps:
Execute this command

```bash
curl --location --request GET 'http://localhost:59882/api/v2/device/name/Dahua-DH-P20A1-5e76b1b6-2cc0-4289-91ef-30442025d9aa/Snapshot'
```
Then return 
```json
{
    "apiVersion": "v2",
    "message": "failed to send a http request",
    "statusCode": 503
}
```
and the log is this
```log
2022/08/18 10:01:41 http: panic serving 172.25.0.9:36006: interface conversion: interface {} is nil, not *media.GetSnapshotUriResponse
goroutine 4343 [running]:
net/http.(*conn).serve.func1()
        net/http/server.go:1825 +0xbf
panic({0x557c6e2355c0, 0xc0003fed80})
        runtime/panic.go:844 +0x258
net/http.(*timeoutHandler).ServeHTTP(0xc0003841c0, {0x557c6e2d6c18, 0xc0004a41c0}, 0xc0006b6a00)
        net/http/server.go:3379 +0x94c
github.com/gorilla/mux.(*Router).ServeHTTP(0xc00031dbc0, {0x557c6e2d6c18, 0xc0004a41c0}, 0xc0006b6800)
        github.com/gorilla/mux@v1.8.0/mux.go:210 +0x1cf
net/http.serverHandler.ServeHTTP({0xc0007860c0?}, {0x557c6e2d6c18, 0xc0004a41c0}, 0xc0006b6800)
        net/http/server.go:2916 +0x43b
net/http.(*conn).serve(0xc00031a500, {0x557c6e2d6f60, 0xc00045df80})
        net/http/server.go:1966 +0x5d7
created by net/http.(*Server).Serve
        net/http/server.go:3071 +0x4db
```

If I use ONVIF Device Test Tool 
I can get response `This optional method is not implemented`

 

## 🌍  Your Environment
**Deployment Environment:**
Docker + EdgeX-Compose

**EdgeX Version [**REQUIRED**]:**
2.2
**Anything else relevant?**

lenny-intel, 2022-08-18:
@hahattan , can you look into this issue?

cloudxxx8, 2022-08-19:
I am assigning it to @weichou1229 
please see why it caused panic

weichou1229, 2022-08-26:
@447264188 Could you provide the SOAP response and HTTP status code?

weichou1229, 2022-08-29:
@447264188 Your camera might not support ONVIF GetSnapshotUrl function, I opened the PR to fix the nill error https://github.com/edgexfoundry/device-onvif-camera/pull/146.

PR #146 (2022-08-30): fix: Check whether the onvif response is valid
Check whether the response is valid for the specified ONVIF function

Fix #138

**If your build fails**  due to your commit message not passing the build checks, please review the guidelines here: https://github.com/edgexfoundry/edgex-go/blob/main/.github/Contributing.md

## PR Checklist
Please check if your PR fulfills the following requirements:

- [x] I am not introducing a breaking change (if you are, flag in conventional commit message with `BREAKING CHANGE:` describing the break)
- [x] I am not introducing a new dependency (add notes below i [... cut]
commit dc3ccaf1 (2022-08-30): fix: Check whether the onvif response is valid (#146)

Check whether the response is valid for the specified ONVIF function

Fix #138
closed 2022-08-30 by lenny-intel