edgexfoundry/device-onvif-camera #140: 'CreateRules' API can not working

447264188, 2022-08-23:
### Summary of the problem:
'CreateRules' API can not working

### Issue reproducible on below camera models:
HIKVISION-DS-2CD3326WDV3-I
HIKVISION-DS-2CD3T27FWDA4-LS

### Tester
from ThunderSoft

### Reproduce steps:
Execute this command

``` bash 
curl --location --request PUT 'http://10.0.20.42:59882/api/v2/device/name/HIKVISION-DS-2CD3T27FWDA4-LS-9b7f0000-0d7d-11b3-8131-40acbf7e7527/CreateRules' \
--header 'Content-Type: application/json' \
--data-raw '{
    "CreateRules": {
        "ConfigurationToken": "VideoAnalyticsToken",
        "Rule": [
            {
                "Name": "MyFieldDetector1",
                "Parameters": {
                    "ElementItem": [
                        {
                            "Name": "Field"
                        }
                    ]
                },
                "Type": "tt:FieldDetector"
            }
        ]
    }
}'
```
Then return

``` json
{
    "apiVersion": "v2",
    "statusCode": 200
}
```
Then verify with GetRules command-

``` bash 
curl --location --request GET 'http://10.0.20.42:59882/api/v2/device/name/HIKVISION-DS-2CD3T27FWDA4-LS-9b7f0000-0d7d-11b3-8131-40acbf7e7527/Rules?jsonObject=eyJDb25maWd1cmF0aW9uVG9rZW4iOiJWaWRlb0FuYWx5dGljc1Rva2VuIn0='
```
I can't get the rules I added

## 🌍  Your Environment
**Deployment Environment:**
Docker + EdgeX-Compose

**EdgeX Version [**REQUIRED**]:**
2.2
**Anything else relevant?**

cloudxxx8, 2022-08-23:
do you have the error log?
could you please try to call the Camera 'CreateRules' API directly to make sure your cameras support his API correctly?

447264188, 2022-09-30:
this HIKVISION camera does not support  CreateRules by onvif

weichou1229, 2022-09-30:
There is a page to indicate some cameras' ONVIF function does not work or is not implemented.
In your case, the camera might just return a 200 HTTP status code and doesn't do anything.

447264188, 2022-09-30:
Thank you very much for your answer

lenny-intel, 2023-01-09:
Closing as answered

closed 2023-01-09 by lenny-intel