edgexfoundry/device-onvif-camera #142: 'ModifyRules' API can not working

447264188, 2022-08-23:
### Summary of the problem:
'ModifyRules' API can not working

### Issue reproducible on below camera models:
HIKVISION-DS-2CD3326WDV3-I
HIKVISION-DS-2CD3T27FWDA4-LS

### Tester
from ThunderSoft

### Reproduce steps:
Execute this command

``` bash 
curl --location --request PUT 'http://10.0.20.42:59882/api/v2/device/name/HIKVISION-DS-2CD3T27FWDA4-LS-9b7f0000-0d7d-11b3-8131-40acbf7e7527/Rules' \
--header 'Content-Type: application/json' \
--data-raw '{
    "Rules": {
        "ConfigurationToken": "VideoAnalyticsToken",
        "Rule": [
            {
                "Name": "MyMotionDetectorRule",
                "Type": "tt:CellMotionDetector",
                "Parameters": {
                    "SimpleItem": [
                        {
                            "Name": "MinCount",
                            "Value": "4"
                        },
                        {
                            "Name": "AlarmOnDelay",
                            "Value": "1001"
                        },
                        {
                            "Name": "AlarmOffDelay",
                            "Value": "1002"
                        },
                        {
                            "Name": "ActiveCells",
                            "Value": "0P8A8A=="
                        }
                    ]
                }
            }
        ]
    }
}'
```
Then return

``` json
{
    "apiVersion": "v2",
    "statusCode": 200
}
```
Then verify with GetRules command-

``` bash 
curl --location --request GET 'http://10.0.20.42:59882/api/v2/device/name/HIKVISION-DS-2CD3T27FWDA4-LS-9b7f0000-0d7d-11b3-8131-40acbf7e7527/Rules?jsonObject=eyJDb25maWd1cmF0aW9uVG9rZW4iOiJWaWRlb0FuYWx5dGljc1Rva2VuIn0='
```
then return 
``` json
{
    "apiVersion": "v2",
    "statusCode": 200,
    "event": {
        "apiVersion": "v2",
        "id": "eccd65b9-b343-4596-be7a-2249c996bd87",
        "deviceName": "HIKVISION-DS-2CD3T27FWDA4-LS-9b7f0000-0d7d-11b3-8131-40acbf7e7527",
        "profileName": "onvif-camera",
        "sourceName": "Rules",
        "origin": 1661253375362000227,
        "readings": [
            {
                "id": "647cc120-309f-4ac2-8520-0783d102455c",
                "origin": 1661253375362000227,
                "deviceName": "HIKVISION-DS-2CD3T27FWDA4-LS-9b7f0000-0d7d-11b3-8131-40acbf7e7527",
                "resourceName": "Rules",
                "profileName": "onvif-camera",
                "valueType": "Object",
                "value": "",
                "objectValue": {
                    "Rule": [
                        {
                            "Name": "MyMotionDetectorRule",
                            "Parameters": {
                                "SimpleItem": [
                                    {
                                        "Name": "MinCount",
                                        "Value": "5"
                                    },
                                    {
                                        "Name": "AlarmOnDelay",
                                        "Value": "1000"
                                    },
                                    {
                                        "Name": "AlarmOffDelay",
                                        "Value": "1000"
                                    },
                                    {
                                        "Name": "ActiveCells",
                                        "Value": "0P8A8A=="
                                    }
                                ]
                            },
                            "Type": "tt:CellMotionDetector"
                        },
                        {
                            "Name": "MyTamperDetectorRule",
                            "Parameters": {
                                "ElementItem": [
                                    {
                                        "Name": "Field"
                                    }
                                ]
                            },
                            "Type": "hikxsd:TamperDetector"
                        }
                    ]
                }
            }
        ]
    }
}
```
I can't get the rules I update

## 🌍  Your Environment
**Deployment Environment:**
Docker + EdgeX-Compose

**EdgeX Version [**REQUIRED**]:**
2.2
**Anything else relevant?**

cloudxxx8, 2022-08-23:
do you have the error log?
could you please try to call the Camera 'ModifyRules' API directly to make sure your cameras support his API correctly?

lenny-intel, 2023-01-09:
Not supported on Hikvision. https://github.com/edgexfoundry/device-onvif-camera/blob/main/doc/ONVIF-protocol.md#rule-configuration
Closing as answered

closed 2023-01-09 by lenny-intel