edgexfoundry/device-onvif-camera #157: Docs: Tapo C200 SetSystemDateTime invalid

ajcasagrande, 2022-09-08:
# 📚 Docs or Wiki Bug Report

### Description [**REQUIRED**]

It is marked as not working, but it works, but with caveats.

## 🔬 Minimal Reproduction

### What's the affected URL? [**REQUIRED**]
 https://docs.edgexfoundry.org/...

### Expected vs Actual Behavior

## 📷Screenshot 

## 🔥 Exception or Error
<pre><code>

</code></pre>

## 🌍  Your Environment

ajcasagrande, 2022-10-05:
@vyshali-chitikeshi needs to be revalidated after #168

vyshali-chitikeshi, 2022-10-28:
Verified that footnote added on Validation matrix for Tapo Set system date and time and documentation updated correctly.
https://github.com/edgexfoundry/device-onvif-camera/blob/main/doc/onvif-footnotes.md#tapo-c200---setsystemdateandtime