edgexfoundry/device-onvif-camera #158: Date field is missing from GetSystemDateAndTime

ajcasagrande, 2022-09-08:
The `Date` struct is missing the `Date` field containing the actual day of the month.
See https://github.com/IOTechSystems/onvif/issues/18

```json
{
    "apiVersion": "v2",
    "statusCode": 200,
    "event": {
        "apiVersion": "v2",
        "id": "da6f552e-e4b3-4d75-a2ce-989745b8ef7b",
        "deviceName": "TP-Link-C200-3fa1fe68-b915-4053-a3e1-1027f5ea88f4",
        "profileName": "onvif-camera",
        "sourceName": "SystemDateAndTime",
        "origin": 1659672865881317546,
        "readings": [
            {
                "id": "aa61a541-beb5-4be0-bb7a-625ce60a980e",
                "origin": 1659672865881317546,
                "deviceName": "TP-Link-C200-3fa1fe68-b915-4053-a3e1-1027f5ea88f4",
                "resourceName": "SystemDateAndTime",
                "profileName": "onvif-camera",
                "valueType": "Object",
                "value": "",
                "objectValue": {
                    "SystemDateAndTime": {
                        "DateTimeType": "NTP",
                        "DaylightSavings": true,
                        "Extension": "",
                        "LocalDateTime": {
                            "Date": {
                                "Month": "8",
                                "Year": "2022"
                            },
                            "Time": {
                                "Hour": "21",
                                "Minute": "14",
                                "Second": "25"
                            }
                        },
                        "TimeZone": {
                            "TZ": "GMT+08:00DST+07:00,M3.2.0/02:00:00,M11.1.0/02:00:00\n"
                        },
                        "UTCDateTime": {
                            "Date": {
                                "Month": "8",
                                "Year": "2022"
                            },
                            "Time": {
                                "Hour": "4",
                                "Minute": "14",
                                "Second": "25"
                            }
                        }
                    }
                }
            }
        ]
    }
}
```

ajcasagrande, 2022-10-05:
Should be fixed via #168. 

@vyshali-chitikeshi Needs to be verified.

ajcasagrande, 2022-10-12:
Confirmed fixed via integration tests.

closed 2022-10-12 by ajcasagrande

vyshali-chitikeshi, 2022-10-28:
Verified using images on 10/27/22 manually on tapo camera and it does display date field correctly on tapo camera with Get system date and time.

{
    "apiVersion": "v2",
    "statusCode": 200,
    "event": {
        "apiVersion": "v2",
        "id": "ae413ac2-ca97-4083-b646-7a4c9d84a5da",
        "deviceName": "TP-Link-C200-3fa1fe68-b915-4053-a3e1-1027f5ea8a87",
        "profileName": "onvif-camera",
        "sourceName": "SystemDateAndTime",
        "origin": 1666985367449660574,
        "readings": [
            {
                "id": "bf17b447-3791-4067-99a9-55a4f6744555",
                "origin": 1666985367449660574,
                "deviceName": "TP-Link-C200-3fa1fe68-b915-4053-a3e1-1027f5ea8a87",
                "resourceName": "SystemDateAndTime",
                "profileName": "onvif-camera",
                "valueType": "Object",
                "value": "",
                "objectValue": {
                    "SystemDateAndTime": {
                        "DateTimeType": "NTP",
                        "DaylightSavings": true,
                        "Extension": "",
                        "LocalDateTime": {
                            "Date": {
                                "Day": 28,
                                "Month": 10,
                                "Year": 2022
                            },
                            "Time": {
                                "Hour": 12,
                                "Minute": 29,
                                "Second": 27
                            }
                        },
                        "TimeZone": {
                            "TZ": "GMT+07:00\n"
                        },
                        "UTCDateTime": {
                            "Date": {
                                "Day": 28,
                                "Month": 10,
                                "Year": 2022
                            },
                            "Time": {
                                "Hour": 19,
                                "Minute": 29,
                                "Second": 27
                            }
                        }
                    }
                }
            }
        ]
    }