edgexfoundry/device-onvif-camera #167: 'BaseNotificationSubscription' and  'PullPointSubscription' API can not working properly

447264188, 2022-09-30:
### Summary of the problem:
'BaseNotificationSubscription' and  'PullPointSubscription' API can not working properly

###  Issue reproducible on below camera models:
Dahua-DH-P20T1

### Tester
from ThunderSoft

### Reproduce steps:
Execute this command

```bash
curl --location --request PUT 'http://10.0.20.42:59882/api/v2/device/name/Dahua-DH-P20T1-3557472d-01be-4a5d-ad9c-5d89120a6473/PullPointSubscription' \
--header 'Content-Type: application/json' \
--data-raw '{
    "PullPointSubscription": {
        "MessageContentFilter": "boolean(//tt:SimpleItem[@Name=\"Rule\"])",
        "InitialTerminationTime": "PT180S",
        "MessageTimeout": "PT180S"
    }
}'

curl --location --request PUT 'http://10.0.20.42:59882/api/v2/device/name/Dahua-DH-P20T1-3557472d-01be-4a5d-ad9c-5d89120a6473/BaseNotificationSubscription' \
--header 'Content-Type: application/json' \
--data-raw '{
    "BaseNotificationSubscription": {
        "TopicFilter": "tns1:RuleEngine/TamperDetector/Tamper",
        "InitialTerminationTime": "PT180S"
    }
}'
```
Then both return 
```json
{
    "apiVersion": "v2",
    "message":"request failed, status code: 500, err: {\"apiVersion\":\"v2\",\"message\":\"error writing DeviceResourece BaseNotificationSubscription for Dahua-DH-P20T1-3557472d-01be-4a5d-ad9c-5d89120a6473 -\> failed to execute write command, \<nil\> -\> failed to create commandValue for the web service 'EdgeX' function 'SubscribeCameraEvent' -\> failed to create the BaseNotification for resource 'BaseNotificationSubscription' -\> invalid SubscribeResponse of type \<nil\> for the camera Dahua-DH-P20T1-3557472d-01be-4a5d-ad9c-5d89120a6473\",\"statusCode\":500}",
    "statusCode": 500
}
```
and the log is this
```log
level=ERROR ts=2022-09-30T06:13:33.847260132Z app=device-onvif-camera source=restrouter.go:167 X-Correlation-ID=b9ea9a51-6aa1-4cdc-a3db-5889f757efda msg="error writing DeviceResourece PullPointSubscription for Dahua-DH-P20T1-3557472d-01be-4a5d-ad9c-5d89120a6473 -> failed to execute write command, <nil> -> failed to create commandValue for the web service 'EdgeX' function 'SubscribeCameraEvent' -> failed to create the PullPoint subscription for resource 'PullPointSubscription' -> invalid CreatePullPointSubscriptionResponse of type <nil> for the camera Dahua-DH-P20T1-3557472d-01be-4a5d-ad9c-5d89120a6473"

level=ERROR ts=2022-09-30T06:13:38.655879003Z app=device-onvif-camera source=restrouter.go:167 X-Correlation-ID=588fa612-9f78-4872-be96-f2468a04c08c msg="error writing DeviceResourece BaseNotificationSubscription for Dahua-DH-P20T1-3557472d-01be-4a5d-ad9c-5d89120a6473 -> failed to execute write command, <nil> -> failed to create commandValue for the web service 'EdgeX' function 'SubscribeCameraEvent' -> failed to create the BaseNotification for resource 'BaseNotificationSubscription' -> invalid SubscribeResponse of type <nil> for the camera Dahua-DH-P20T1-3557472d-01be-4a5d-ad9c-5d89120a6473"

```

## 🌍  Your Environment
**Deployment Environment:**
Docker + EdgeX-Compose

**EdgeX Version [**REQUIRED**]:**
2.2
**Anything else relevant?**

weichou1229, 2022-09-30:
Could you enable the debug log or you can try the [Postman](https://github.com/edgexfoundry/device-onvif-camera/blob/main/doc/test-with-postman.md) to test the camera directly?

447264188, 2022-09-30:
pullpointsubscriber.go    133 lines

subscriptionResponse, ok := respContent.(*event.CreatePullPointSubscriptionResponse)

respContent is <nil>

weichou1229, 2022-09-30:
Could you enable the debug logs? This will show the SOAP request and response. Or test with Postman to send the ONVIF request.

lindseysimple, 2023-12-12:
Since this issue has no activity since 2022, closing the issue.

closed 2023-12-12 by lindseysimple