edgexfoundry/device-onvif-camera #173: Inconsistent naming of Onvif functions to EdgeX commands

ajcasagrande, 2022-10-05:
Currently the service has inconsistent names for EdgeX commands compared to the onvif equivalent. For example, some commands remove the `Get` such as the `StreamUri` command, however others such as `GetPresets` keep the `Get` in the beginning.

Here is a list of all the EdgeX commands which begin with `Get` or `Set`:

```
SetSystemFactoryDefault
GetVideoEncoderConfigurations
GetNodes
GetNode
GetConfigurations
GetConfigurationOptions
GetStatus
SetPreset
GetPresets
SetHomePosition
GetEventProperties
GetAnalyticsConfigurations
GetSupportedAnalyticsModules
GetAnalyticsModuleOptions
GetSupportedRules
GetRuleOptions
```

Obtained via the command:
```shell
cat cmd/res/profiles/camera.yaml | sed -En 's/.*name: "((Get|Set)[^"]+)".*/\1/p'
```

Here is a scraped table with all of the EdgeX commands and their Onvif equivalents.

| REST | EdgeX Command | Onvif Call | Difference |
| ---------|-------------------------|---------------|-------------- |
| GET | Hostname | GetHostname | Get |
| PUT | Hostname | SetHostname | Set |
| GET | DNS | GetDNS | Get |
| PUT | DNS | SetDNS | Set |
| GET | NetworkInterfaces | GetNetworkInterfaces | Get |
| PUT | NetworkInterfaces | SetNetworkInterfaces | Set |
| GET | NetworkProtocols | GetNetworkProtocols | Get |
| PUT | NetworkProtocols | SetNetworkProtocols | Set |
| GET | NetworkDefaultGateway | GetNetworkDefaultGateway | Get |
| PUT | NetworkDefaultGateway | SetNetworkDefaultGateway | Set |
| GET | DeviceInformation | GetDeviceInformation | Get |
| GET | SystemDateAndTime | GetSystemDateAndTime | Get |
| PUT | SystemDateAndTime | SetSystemDateAndTime | Set |
| PUT | SetSystemFactoryDefault | SetSystemFactoryDefault |  |
| PUT | SystemReboot | SystemReboot |  |
| GET | MetadataConfigurations | GetMetadataConfigurations | Get |
| GET | MetadataConfiguration | GetMetadataConfiguration | Get |
| PUT | MetadataConfiguration | SetMetadataConfiguration | Set |
| GET | CompatibleMetadataConfigurations | GetCompatibleMetadataConfigurations | Get |
| GET | MetadataConfigurationOptions | GetMetadataConfigurationOptions | Get |
| PUT | AddMetadataConfiguration | AddMetadataConfiguration |  |
| PUT | RemoveMetadataConfiguration | RemoveMetadataConfiguration |  |
| GET | Profiles | GetProfiles | Get |
| GET | StreamUri | GetStreamUri | Get |
| GET | GetVideoEncoderConfigurations | GetVideoEncoderConfigurations |  |
| GET | VideoEncoderConfiguration | GetVideoEncoderConfiguration | Get |
| PUT | VideoEncoderConfiguration | SetVideoEncoderConfiguration | Set |
| GET | VideoEncoderConfigurationOptions | GetVideoEncoderConfigurationOptions | Get |
| GET | Users | GetUsers | Get |
| PUT | Users | SetUser | SetUser |
| PUT | CreateUsers | CreateUsers |  |
| PUT | DeleteUsers | DeleteUsers |  |
| GET | DiscoveryMode | GetDiscoveryMode | Get |
| PUT | DiscoveryMode | SetDiscoveryMode | Set |
| PUT | AddScopes | AddScopes |  |
| GET | Scopes | GetScopes | Get |
| PUT | Scopes | SetScopes | Set |
| PUT | RemoveScopes | RemoveScopes |  |
| GET | GetNodes | GetNodes |  |
| GET | GetNode | GetNode |  |
| GET | GetConfigurations | GetConfigurations |  |
| GET | Configuration | GetConfiguration | Get |
| PUT | Configuration | SetConfiguration | Set |
| GET | GetConfigurationOptions | GetConfigurationOptions |  |
| PUT | AddPTZConfiguration | AddPTZConfiguration |  |
| PUT | RemovePTZConfiguration | RemovePTZConfiguration |  |
| PUT | AbsoluteMove | AbsoluteMove |  |
| PUT | RelativeMove | RelativeMove |  |
| PUT | ContinuousMove | ContinuousMove |  |
| GET | GetStatus | GetStatus |  |
| PUT | Stop | Stop |  |
| PUT | SetPreset | SetPreset |  |
| GET | GetPresets | GetPresets |  |
| PUT | GotoPreset | GotoPreset |  |
| PUT | RemovePreset | RemovePreset |  |
| PUT | GotoHomePosition | GotoHomePosition |  |
| PUT | SetHomePosition | SetHomePosition |  |
| PUT | SendAuxiliaryCommand | SendAuxiliaryCommand |  |
| GET | GetEventProperties | GetEventProperties |  |
| GET | Media2Profiles | GetProfiles | GetProfiles |
| GET | GetAnalyticsConfigurations | GetAnalyticsConfigurations |  |
| PUT | AddConfiguration | AddConfiguration |  |
| PUT | RemoveConfiguration | RemoveConfiguration |  |
| GET | GetSupportedAnalyticsModules | GetSupportedAnalyticsModules |  |
| GET | AnalyticsModules | GetAnalyticsModules | Get |
| PUT | AnalyticsModules | ModifyAnalyticsModules | Modify |
| PUT | CreateAnalyticsModules | CreateAnalyticsModules |  |
| PUT | DeleteAnalyticsModules | DeleteAnalyticsModules |  |
| GET | GetAnalyticsModuleOptions | GetAnalyticsModuleOptions |  |
| GET | GetSupportedRules | GetSupportedRules |  |
| GET | Rules | GetRules | Get |
| PUT | Rules | ModifyRules | Modify |
| GET | GetRuleOptions | GetRuleOptions |  |
| PUT | CreateRules | CreateRules |  |
| PUT | DeleteRules | DeleteRules |  |

cloudxxx8, 2022-10-17:
it is kind of easy fix in the default profile.  We don't hardcode it in the source code.  We can add another GetXXX in the default profile and make them backward consistent. 
https://github.com/edgexfoundry/device-onvif-camera/blob/7ec405a7ebe2d3fd94ca179867e83cca09fb31d0/cmd/res/profiles/camera.yaml#L206-L233

ajcasagrande, 2022-10-21:
@cloudxxx8 @lenny-intel @trleasher-intel alternatively we can add secondary functions for the ones that still have Get and Set in the name (like for `GetSupportedRules`, add a `SupportedRules` duplicate). I'm not sure which is better, I just think it should be consistent whichever one we go with (right now some have the Get and some remove the Get).

We can never truly be word for word the same as the ONVIF spec, as there are conflicting names across services (which is why `Media2Profiles` exists).

cloudxxx8, 2022-10-24:
I am fine with either way.  It's good to make the names as consistent as possible.

lenny-intel, 2023-04-03:
@ajcasagrande, @presatish , add a task for this for EdgeX US?

closed 2023-05-10 by lenny-intel