edgexfoundry/device-onvif-camera #181: Failed to invoke device service device-onvif-camera  call back from core-metadata when tried  to discovery onvif camera

edgego, 2022-10-22:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]

 The issue is located in: 

Metadata

### Is this a regression?

 Yes, the previous version in which this bug was not present was: ....
No

### Description and Minimal Reproduction [**REQUIRED**]

core-metadata  version 2.2.0
device-onvif-camera version 2.2.0

tried to discovery onvif camera with device service device-onvif-camera, checked its log get error:

level=INFO ts=2022-10-20T07:05:05.83079411Z app=device-onvif-camera source=async.go:104 msg="Adding discovered device HIKVISION-DS-2DE2402IW-D3-W-XM-617dc000-4574-11b4-82be-2428fdc5062f to Metadata"
level=ERROR ts=2022-10-20T07:05:10.832370558Z app=device-onvif-camera source=async.go:120 msg="failed to create discovered device HIKVISION-DS-2DE2402IW-D3-W-XM-617dc000-4574-11b4-82be-2428fdc5062f: request failed, status code: 503, err: HTTP request timeout"

then cheked core-metadata log:

level=ERROR ts=2022-10-22T12:43:29.790741212Z app=core-metadata source=notify.go:238 msg="fail to invoke device service callback for updating device service device-onvif-camera, err: failed to send a http request -> Put \"http://device-onvif-camera:59984/api/v2/callback/service\": dial tcp 10.43.218.60:59984: i/o timeout"
level=WARN ts=2022-10-22T12:43:31.960365751Z app=core-metadata source=notify.go:80 msg="Skipping device validation for device HIKVISION-DS-2DE2402IW-D3-W-XM-617dc000-4574-11b4-82be-2428fdc5062f (device service device-onvif-camera unavailable)"

## 🔥 Exception or Error
<pre><code>

</code></pre>

## 🌍  Your Environment
**Deployment Environment:**

**EdgeX Version [**REQUIRED**]:**

**Anything else relevant?**

lenny-intel, 2022-10-22:
@edgego , obviously `device-onvif-camera:59984` is not responding in the docker network for Core Metadata. Are you running the device service from command-line (hybrid) or in Docker?
Looks like the first time you ran the device service it was in docker and registered with host name `device-onvif-camera` which is a docker network name.

If in fact you are in hybrid mode, do the following:
1. Stop Edgex services running in Docker
2. Remove volumes so old Host name is removed from Core Metadata
3. Restart docker services w/o `device-onvif-camera` (otherwise you will recreate same issue)
4. Set `Service.Host` in local `device-onvif-camera` `configuration.toml` to IP address of host system
5. Run `device-onvif-camera from command-line

See https://docs.edgexfoundry.org/2.2/getting-started/Ch-GettingStartedHybrid/ for more details about working in a hybrid deployment.

lenny-intel, 2023-01-25:
closing due to no activity

closed 2023-01-25 by lenny-intel