edgexfoundry/device-onvif-camera #211: Hikvision-DS-2DE2A404IW-DE3- Get device capabilities gives an error

vyshali-chitikeshi, 2023-01-13:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]

 The issue is located in: 
Onvif

### Is this a regression?
No

 Yes, the previous version in which this bug was not present was: ....

### Description and Minimal Reproduction [**REQUIRED**]
Get device capabilities API gives an error on  Hikvision DS-2DE2A404IW-DE3 camera

## 🔥 Exception or Error
<pre><code>

{
    "apiVersion": "v2",
    "message":"request failed, status code: 500, err: {\"apiVersion\":\"v2\",\"message\":\"error reading DeviceResourece Capabilities for HIKVISION-DS-2DE2A404IW-DE3-099f4000-4d50-11b4-82c8-c06ded544d95 -\> failed to execute read command -\> failed to execute the request for the function 'GetCapabilities' of web service 'Device'. Onvif error: fault reason: the parameter value is illegal, fault detail: , fault code: env:Sender ter:InvalidArgVal ter:InvalidParameter\",\"statusCode\":500}",
    "statusCode": 500
}
</code></pre>

## 🌍  Your Environment
**Deployment Environment:**

**EdgeX Version [**REQUIRED**]:**

**Anything else relevant?**
attached API command used & 
[image]
output

ajcasagrande, 2023-01-25:
If you look at the official docs: https://www.onvif.org/ver10/device/wsdl/devicemgmt.wsdl#op.GetCapabilities

there is an optional field `Category`, which I believe this camera incorrectly requires. Try using the following body:

```
{
  "Category": ["All"]
}
```

vyshali-chitikeshi, 2023-01-25:
Same error with optional parameter of Category in body. attached screenshot

vyshali-chitikeshi, 2023-01-25:
[image]

ajcasagrande, 2023-03-14:
Fixed with upgrade to onvif lib v0.1.6 from https://github.com/IOTechSystems/onvif/pull/21

closed 2023-03-14 by ajcasagrande