edgexfoundry/device-onvif-camera #212: Hikvision-DS-2DE2A404IW-DE3- Get device service capabilities API gives an error

vyshali-chitikeshi, 2023-01-13:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]

 The issue is located in: 
Onvif service

### Is this a regression?
No

 Yes, the previous version in which this bug was not present was: ....

### Description and Minimal Reproduction [**REQUIRED**]

Device service capabilities API gives an error
## 🔥 Exception or Error
<pre><code>

{
    "apiVersion": "v2",
    "message":"request failed, status code: 500, err: {\"apiVersion\":\"v2\",\"message\":\"error reading DeviceResourece DeviceCapabilities for HIKVISION-DS-2DE2A404IW-DE3-099f4000-4d50-11b4-82c8-c06ded544d95 -\> failed to execute read command -\> failed to create 'GetServiceCapabilities' response for the web service 'Device' -\> cannot unmarshal into device.EAPMethodTypes\",\"statusCode\":500}",
    "statusCode": 500
}
[image]

</code></pre>

## 🌍  Your Environment
**Deployment Environment:**

**EdgeX Version [**REQUIRED**]:**

**Anything else relevant?**

vyshali-chitikeshi, 2023-01-13:
[image]

ajcasagrande, 2023-01-25:
Looks like an issue with the internal onvif library

lenny-intel, 2023-01-25:
@ajcasagrande , bumped it to a Medium. Can you please submit bug against the IOTech package and link it here?

vyshali-chitikeshi, 2023-06-07:
It works correctly with latest V3 code, find attached.
Closing this defect
[image]

closed 2023-06-07 by vyshali-chitikeshi