edgexfoundry/device-onvif-camera #213: Hikvision-DS-2DE2A404IW-DE3- Set scopes API gives an error

vyshali-chitikeshi, 2023-01-13:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]

 The issue is located in: 

Onvif service
### Is this a regression?
No

 Yes, the previous version in which this bug was not present was: ....

### Description and Minimal Reproduction [**REQUIRED**]
Set scopes API supposed to update all configurable scopes with desired scope item, however it throws an error. API command and output attached.

## 🔥 Exception or Error
<pre><code>

{
    "apiVersion": "v2",
    "message":"request failed, status code: 500, err: {\"apiVersion\":\"v2\",\"message\":\"error writing DeviceResourece Scopes for HIKVISION-DS-2DE2A404IW-DE3-099f4000-4d50-11b4-82c8-c06ded544d95 -\> failed to execute write command, \<nil\> -\> failed to execute the request for the function 'SetScopes' of web service 'Device'. Onvif error: fault reason: the parameter value is illegal, fault detail: User NULL, fault code: env:Sender ter:InvalidArgVal ter:InvalidParameter\",\"statusCode\":500}",
    "statusCode": 500
}
[image]

</code></pre>

## 🌍  Your Environment
**Deployment Environment:**

**EdgeX Version [**REQUIRED**]:**

**Anything else relevant?**

vyshali-chitikeshi, 2023-01-13:
[image]

ajcasagrande, 2023-01-25:
@vyshali-chitikeshi According to our documentation, the body is supposed to look different than you have it. Try this:

```json
{
  "Scopes": {
    "Scopes": [
      "http//:123"
    ]
  }
}
```

vyshali-chitikeshi, 2023-01-25:
@ajcasagrande Not sure why I used ScopeItem, i think that's  how we have been testing on Tapo as well.
With API body you provided it works correctly on Hikvision camera. Closing this defect .

closed 2023-01-25 by vyshali-chitikeshi