edgexfoundry/device-onvif-camera #214: Hikvision-DS-2DE2A404IW-DE3 camera- Configuration of Analytics profile - Add Configuration API not working

vyshali-chitikeshi, 2023-01-13:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]

 The issue is located in: 

Device onvif
### Is this a regression?

 Yes, the previous version in which this bug was not present was: ....

No
### Description and Minimal Reproduction [**REQUIRED**]

Adding analytics configuration to media2 profile fails on Hikvision camera. Gives invalid parameter error even with valid token value.
attached api output.
## 🔥 Exception or Error
<pre><code>

{
    "apiVersion": "v2",
    "message":"request failed, status code: 500, err: {\"apiVersion\":\"v2\",\"message\":\"error writing DeviceResourece AddConfiguration for HIKVISION-DS-2DE2A404IW-DE3-099f4000-4d50-11b4-82c8-c06ded544d95 -\> failed to execute write command, \<nil\> -\> failed to execute the request for the function 'AddConfiguration' of web service 'Media2'. Onvif error: fault reason: the parameter value is illegal, fault detail: , fault code: env:Sender ter:InvalidArgVal ter:InvalidParameter\",\"statusCode\":500}",
    "statusCode": 500
}
</code></pre>

## 🌍  Your Environment
**Deployment Environment:**

**EdgeX Version [**REQUIRED**]:**

**Anything else relevant?**

vyshali-chitikeshi, 2023-01-13:
[image]

ajcasagrande, 2023-01-23:
@vyshali-chitikeshi try removing the `s` from `Configurations`. The documentation lists it as `Configuration`.

[image]
[image]

https://app.swaggerhub.com/apis-docs/EdgeXFoundry1/device-onvif-camera/2.3.0#/Analytics%20-%20Profile%20Configuration/put_api_v2_device_name__EDGEX_DEVICE_NAME__AddConfiguration

presatish, 2023-01-25:
@ajcasagrande same error even with `Configuration`

vyshali-chitikeshi, 2023-01-25:
I do see same error both with Configuration and Configurations

vyshali-chitikeshi, 2023-06-07:
Issue still exists with latest V3 code, find attached
[image]