edgexfoundry/device-onvif-camera #240: onvif camera + example app service

Einse57, 2023-02-25:
# 📚 Docs or Wiki Bug Report

### Description [**REQUIRED**]
Issue 1:
Following https://github.com/edgexfoundry/device-onvif-camera/blob/main/doc/guides/SimpleStartupGuide.md and https://github.com/edgexfoundry/device-onvif-camera/blob/main/doc/using-demo-app.md on a fresh installation currently produces version mismatch at "Configure and Build the App - Run the app" step

Issue 2: Docs at https://github.com/edgexfoundry/device-onvif-camera/blob/main/doc/using-demo-app.md assume Go is installed and dont call it out as a dependency

## 🔥 Exception or Error
<pre><code>
/application-services/custom/camera-management# make run-app
EDGEX_SECURITY_SECRET_STORE=false ./app-camera-management
level=INFO ts=2023-02-25T18:37:55.2111707Z app=app-camera-management source=service.go:496 msg="Starting app-camera-management 0.0.0 "
level=INFO ts=2023-02-25T18:37:55.2186979Z app=app-camera-management source=config.go:392 msg="Loaded service configuration from ./res/configuration.toml"
level=INFO ts=2023-02-25T18:37:55.2246103Z app=app-camera-management source=config.go:556 msg="Using local configuration from file (0 envVars overrides applied)"
level=INFO ts=2023-02-25T18:37:55.2246605Z app=app-camera-management source=clients.go:160 msg="Using REST for 'core-command' clients @ http://localhost:59882"
level=INFO ts=2023-02-25T18:37:55.2247004Z app=app-camera-management source=clients.go:160 msg="Using REST for 'core-metadata' clients @ http://localhost:59881"
level=INFO ts=2023-02-25T18:37:55.2267281Z app=app-camera-management source=telemetry.go:78 msg="Starting CPU Usage Average loop"
level=ERROR ts=2023-02-25T18:37:55.236847Z app=app-camera-management source=version.go:137 msg="Core Services version (3.0.0-dev.67) is not compatible with SDK's version(v2.3.0)"
level=ERROR ts=2023-02-25T18:37:55.2369082Z app=app-camera-management source=factory.go:43 msg="App Service initialization failed: boostrapping failed"
error: unable to create new app service app-camera-management!
level=INFO ts=2023-02-25T18:37:55.2369253Z app=app-camera-management source=telemetry.go:83 msg="Exiting CPU Usage Average loop"
make: *** [Makefile:29: run-app] Error 255

</code></pre>

## 🌍  Your Environment

lenny-intel, 2023-02-26:
@Einse57 , this is because you are using the example Camera App (built for 2.3.0 release) with the latest WIP dev ONVIF camera code for the upcoming 3.0.0 release. Please use the code from the ONVIF camera 2.3.0 tag, which is compatible with the example Camera App, here: https://github.com/edgexfoundry/device-onvif-camera/tree/v2.3.0

ajcasagrande, 2023-02-27:
@Einse57 for issue 2, it we have not completely dockerized the example app yet, so that is why installing Go is required. You can install the latest version here: https://go.dev/doc/install

Einse57, 2023-03-01:
I removed the newer image that was auto pulled for device-onvif-camera then went back and checked out the v2.3.0 tag from https://github.com/edgexfoundry/device-onvif-camera.git then ran make docker successfully. I see the corresponding dev 0.0.0 image from my build of the onvif camera device service but when I rerun it's still throwing the following - Do I need to make sure all services are v2.3.0? Is there an easy way to do that in the compose framework?

level=INFO ts=2023-02-27T22:03:50.7634665Z app=app-camera-management source=telemetry.go:78 msg="Starting CPU Usage Average loop"
level=ERROR ts=2023-02-27T22:03:50.7759503Z app=app-camera-management source=version.go:137 msg="Core Services version (3.0.0-dev.67) is not compatible with SDK's version(v2.3.0)"
level=ERROR ts=2023-02-27T22:03:50.7760047Z app=app-camera-management source=factory.go:43 msg="App Service initialization failed: boostrapping failed"
error: unable to create new app service app-camera-management!
level=INFO ts=2023-02-27T22:03:50.7760372Z app=app-camera-management source=telemetry.go:83 msg="Exiting CPU Usage Average loop"

lenny-intel, 2023-03-01:
Yes, the camera management app in examples is for 2.x version only. You are running the latest 3.0.0 WIP services from edgex-compose. Simply checkout the levski branch of edgex-compose and run the compose file from that branch.

Einse57, 2023-03-01:
It's working! Checking out the v2.3.0 tag in edgex-compose worked. I had assumed that this was already 2.3.0 since that was the latest tag. 

I think where I went wrong was in focusing on the device-onvif-camera repo vs fixing my version mismatch in the compose repo (since the simple tutorial has you pull them vs build them first).

Maybe moving the checkout step of device-onvif-camera could be move later in the guide just before use?

https://github.com/edgexfoundry/device-onvif-camera/blob/main/doc/guides/SimpleStartupGuide.md#deploy-edgex-and-onvif-device-camera-microservice

Einse57, 2023-03-01:
Last q on the tutorial here - I see the menus have a simulated camera option - Is there a fast way to get something like this going in the documentation?

I couldnt find by searching or in the simple setup / camera demo. What's recommended?

lenny-intel, 2023-03-01:
> Last q on the tutorial here - I see the menus have a simulated camera option - Is there a fast way to get something like this going in the documentation?
> 
> I couldnt find by searching or in the simple setup / camera demo. What's recommended?

@ajcasagrande can better answer this.

ajcasagrande, 2023-03-02:
> Last q on the tutorial here - I see the menus have a simulated camera option - Is there a fast way to get something like this going in the documentation?
> 
> I couldnt find by searching or in the simple setup / camera demo. What's recommended?

We have used the [Happytime Multi ONVIF Server](https://www.happytimesoft.com/products/multi-onvif-server/index.html) as a simulator for Onvif cameras. The free version out of the box is limited to 2 cameras, and does not integrate as nicely into the system.

Internally we have purchased the source code to it and have modified it to improve some of that compatibility. We currently use it as one of our devices for our automated integration testing.

Einse57, 2023-03-02:
Awesome, total success! Thanks for the guidance here.

closed 2023-03-06 by presatish