edgexfoundry/device-onvif-camera #256: Get command should return server error instead of timeout error

weichou1229, 2023-03-10:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]

 The issue is located in device-onvif-camera.

### Is this a regression?

 No

### Description and Minimal Reproduction [**REQUIRED**]
The request command throws the timeout error when the ONVIF device is unavailable.

Minimal Reproduction:
- Run core-service and device-onvif-camera
- Add a device for the unavailable camera.
- Send a get command.
   ```
   curl http://localhost:59882/api/v2/device/name/Camera01/NetworkConfiguration
   request timeout
   ```
- The device-onvif-camera throws a timeout error instead of the unavailable error

## 🔥 Exception or Error
```
level=ERROR ts=2023-03-09T05:42:29.769729803Z app=core-command source=http.go:47 X-Correlation-ID=af0529e8-46b2-4481-8c71-e8d3cf1c6e5a msg="request failed, status code: 500, err: {\"apiVersion\":\"v2\",\"message\":\"error reading DeviceCommand NetworkConfiguration for Camera01 -\\u003e failed to initialize onvif client for 'Camera01' camera -\\u003e failed to initialize Onvif device client -\\u003e camera is not available at 192.168.12.123:80 or it does not support ONVIF services\",\"statusCode\":500}"
level=ERROR ts=2023-03-09T05:42:29.769791765Z app=core-command source=encoding.go:32 msg="Error encoding the data: http: Handler timeout"
level=ERROR ts=2023-03-09T05:46:52.767701787Z app=core-command source=http.go:47 X-Correlation-ID=cf1936c1-9f84-43e8-af44-b272b72c88b8 msg="request failed, status code: 503, err: HTTP request timeout"
```

## 🌍  Your Environment
**Deployment Environment:**
MacOS

**EdgeX Version [**REQUIRED**]:**
Main branch

**Anything else relevant?**

ajcasagrande, 2023-03-13:
@weichou1229 seems like a `503 - Service Unavailable` is the right code, but the message is the problem? Also 502 or 504 could potentially be used as well, but 503 probably makes the most sense.

weichou1229, 2023-03-14:
@ajcasagrande 
I thought the problem is we used 5s to create a new Onvif client. The HTTP request already timeout on the service level even if the Onvif client threw the unavailable error.

### service HTTP timeout
```
  Service:
    ...
    RequestTimeout: "5s"
```
https://github.com/edgexfoundry/edgex-go/blob/0d6a49cb75d16cdeffa3d89b53872b828db0197b/cmd/core-common-config-bootstrapper/res/configuration.yaml#L31

### Onvlif client timeout
```
# Custom configs
[AppCustom]
# The number of seconds to wait when making an Onvif request before timing out
RequestTimeout = 5 # Seconds
```
https://github.com/edgexfoundry/device-onvif-camera/blob/33d7f92d5661b99b9b96e2663b29a2bf689d0f01/internal/driver/onvifclient.go#L72-L83

cloudxxx8, 2023-04-14:
@ajcasagrande This issue cause the automation test case failed.  May we set the default AppCustom.RequestTimeout config to **4** (less than the service request tmieout)?  Then, the service can reply the proper error content in the response.  If you agree with this solution, I can assign someone to work on it.  Thanks.

ajcasagrande, 2023-04-20:
@cloudxxx8 yes that is fine

PR #322 (2023-04-24): fix: Get command should return server error instead of timeout error
Set the default AppCustom.RequestTimeout config to 4 (less than the service request tmieout). Then, the service can reply the proper error content in the response.

Close #256

**If your build fails**  due to your commit message not passing the build checks, please review the guidelines here: https://github.com/edgexfoundry/edgex-go/blob/main/.github/Contributing.md

## PR Checklist
Please check if your PR fulfills the following requirements:

- [x] I am not introducing a breaking change (if you are, flag in conventional comm [... cut]
closed 2023-04-24 by cloudxxx8