edgexfoundry/device-onvif-camera #308: Device SDK cache inconsistency, causing device not found errors

ajcasagrande, 2023-04-17:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]
device-sdk-go

### Is this a regression?
Yes

### Description and Minimal Reproduction [**REQUIRED**]
All edgex services are running normally. Called docker stop <container id of onvif service>. Wait a while. Call docker start <same container id>. Observe issue where no device commands work due to 404 device not found error, even though the devices clearly exist in core-metadata.

The first error message code is located in `device.Initialize` and it calls `for _, device := range d.sdkService.Devices() {`, which in turn calls `return cache.Devices().All()`, which leads me to believe the devices do in fact exist in the cache.

## 🔥 Exception or Error
<pre><code>
level=INFO ts=2023-02-17T16:47:20.247530253Z app=device-onvif-camera source=driver.go:141 msg="Initializing onvif client for 'tp-link-Tapo-C200-3fa1fe68-b915-4053-a3e1-1027f5ea88f4' camera"
level=ERROR ts=2023-02-17T16:47:20.440493715Z app=device-onvif-camera source=manageddevices.go:56 msg="failed to find Device tp-link-Tapo-C200-3fa1fe68-b915-4053-a3e1-1027f5ea88f4 in cache"
level=ERROR ts=2023-02-17T16:47:20.440673388Z app=device-onvif-camera source=driver.go:145 msg="failed to initialize onvif client for 'tp-link-Tapo-C200-3fa1fe68-b915-4053-a3e1-1027f5ea88f4' camera, skipping this device."
level=INFO ts=2023-02-17T16:47:20.440714682Z app=device-onvif-camera source=driver.go:141 msg="Initializing onvif client for 'HIKVISION-DS-2DE2A404IW-DE3-099f4000-4d50-11b4-82c8-c06ded544d67' camera"
level=ERROR ts=2023-02-17T16:47:20.491345631Z app=device-onvif-camera source=manageddevices.go:56 msg="failed to find Device HIKVISION-DS-2DE2A404IW-DE3-099f4000-4d50-11b4-82c8-c06ded544d67 in cache"
level=ERROR ts=2023-02-17T16:47:20.491489277Z app=device-onvif-camera source=driver.go:145 msg="failed to initialize onvif client for 'HIKVISION-DS-2DE2A404IW-DE3-099f4000-4d50-11b4-82c8-c06ded544d67' camera, skipping this device."
...
level=INFO ts=2023-02-17T16:47:23.952698178Z app=device-onvif-camera source=driver.go:610 msg="Discovered 2 device(s) in 2.44027692s via netscan."
level=INFO ts=2023-02-17T16:47:23.952931049Z app=device-onvif-camera source=async.go:104 msg="Adding discovered device HIKVISION-DS-2DE2A404IW-DE3-099f4000-4d50-11b4-82c8-c06ded544d67 to Metadata"
level=INFO ts=2023-02-17T16:47:23.958015598Z app=device-onvif-camera source=async.go:104 msg="Adding discovered device tp-link-Tapo-C200-3fa1fe68-b915-4053-a3e1-1027f5ea88f4 to Metadata"
level=ERROR ts=2023-02-17T16:47:34.382891635Z app=device-onvif-camera source=restrouter.go:168 X-Correlation-ID=8b894975-57a7-43d3-b1d6-70a83e510116 msg="device HIKVISION-DS-2DE2A404IW-DE3-099f4000-4d50-11b4-82c8-c06ded544d67 not found"
</code></pre>

## 🌍  Your Environment
**Deployment Environment:**
docker compose 2.x
edgex compose levski branch v2.3.0

**EdgeX Version [**REQUIRED**]:**
2.3.0

**Anything else relevant?**

lenny-intel, 2023-04-17:
Moved this to the ONVIF repo for implementing the Start() API for this service

lenny-intel, 2023-04-17:
Latest SDK requires the Start API() to be implemented.

cloudxxx8, 2023-04-25:
@ajcasagrande do you need my team to help fix this issue in 3.0?

presatish, 2023-05-03:
@cloudxxx8 I think we should be able to. We have prioritized this issue along with others to complete by May 10th.

ajcasagrande, 2023-05-03:
@cloudxxx8 Sorry, my notifications seem to not be working properly, or I set my outlook rules incorrectly. Yes, we will address this. Bruce has already added the stub for it, and I can implement it after my current refactor PR, since it may cause conflicts if done separately.

closed 2023-05-08 by ajcasagrande