edgexfoundry/device-onvif-camera #412: The Hikvision NVR device (model: IDS-7816NX) has enabled Onvif, but the device-onvif-camera cannot be automatically discovered.

watjt, 2023-11-22:
# 🐞 Bug Report

### Affected Services [**REQUIRED**]

 The issue is located in: 
device-onvif-camera

### Is this a regression?

 Yes, the previous version in which this bug was not present was: ....
The Hikvision NVR device (model: IDS-7816NX) has enabled Onvif, but the device-onvif-camera cannot be automatically discovered. It works fine when using the ONVIF Device Manager tool.
[image]

[image]

### Description and Minimal Reproduction [**REQUIRED**]

## 🔥 Exception or Error
<pre><code>

level=DEBUG ts=2023-11-22T02:28:54.190349036Z app=device-onvif-camera source=onvifdiscovery.go:213 msg="172.23.142.26:3702: Response 1 of 1: <?xml version=\"1.0\" encoding=\"utf-8\"?>\r\n<soap:Envelope xmlns:soap=\"http://www.w3.org/2003/05/soap-envelope\"  xmlns:wsa=\"http://schemas.xmlsoap.org/ws/2004/08/addressing\"  xmlns:wsd=\"http://schemas.xmlsoap.org/ws/2005/04/discovery\"  xmlns:dn=\"http://www.onvif.org/ver10/network/wsdl\"  xmlns:tds=\"http://www.onvif.org/ver10/device/wsdl\" >\n<soap:Header>\n<wsa:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/ProbeMatches</wsa:Action>\n<wsa:MessageID>uuid:b7df1a5e-8921-11ee-8228-4d52f5bae128</wsa:MessageID>\n<wsa:RelatesTo>uuid:a1a836b3-5246-432c-9ccc-dc6071e25345</wsa:RelatesTo>\n<wsa:To>http://schemas.xmlsoap.org/ws/2004/08/addressing/role/anonymous</wsa:To>\n<wsd:AppSequence InstanceId=\"1700648840\" MessageNumber=\"70\"/>\n</soap:Header>\n<soap:Body>\n<wsd:ProbeMatches>\n<wsd:ProbeMatch>\n<wsa:EndpointReference>\n<wsa:Address>urn:uuid:484a3534-3234-3337-3238-240f9b30ea59</wsa:Address>\n</wsa:EndpointReference>\n<wsd:Types>dn:NetworkVideoTransmitter tds:Device</wsd:Types>\n<wsd:Scopes>onvif://www.onvif.org/type/video_encoder onvif://www.onvif.org/type/ptz onvif://www.onvif.org/type/audio_encoder onvif://www.onvif.org/hardware/iDS-7816NX-Z2%2FX onvif://www.onvif.org/Profile/Streaming onvif://www.onvif.org/Profile/G onvif://www.onvif.org/location/country/china onvif://www.onvif.org/name/DeepinMind onvif://www.onvif.org/type/Network%20Video%20Storage </wsd:Scopes>\n<wsd:XAddrs>http://172.23.142.26:80/onvif/device_service</wsd:XAddrs>\n<wsd:MetadataVersion>10</wsd:MetadataVersion>\n</wsd:ProbeMatch>\n</wsd:ProbeMatches>\n</soap:Body>\n</soap:Envelope>\n"
Onvif WS-Discovery: Find Xaddr: 172.23.142.26:80          EndpointRefAddress: 484a3534-3234-3337-3238-240f9b30ea59
level=DEBUG ts=2023-11-22T02:28:54.198195292Z app=device-onvif-camera source=onvifdiscovery.go:110 msg="No MAC Address match was found for EndpointRefAddress 484a3534-3234-3337-3238-240f9b30ea59"
level=WARN ts=2023-11-22T02:28:54.198259849Z app=device-onvif-camera source=credentials.go:105 msg="Device 484a3534-3234-3337-3238-240f9b30ea59 is missing MAC Address, using default secret name"
level=WARN ts=2023-11-22T02:28:54.206230834Z app=device-onvif-camera source=onvifdiscovery.go:115 msg="failed to create onvif client for the camera 484a3534-3234-3337-3238-240f9b30ea59, failed to initialize Onvif device client -> camera is not available at 172.23.142.26:80 or it does not support ONVIF services"
level=WARN ts=2023-11-22T02:28:54.206275879Z app=device-onvif-camera source=discover.go:158 msg="issue converting probe result to discovered device: failed to create onvif client for the camera 484a3534-3234-3337-3238-240f9b30ea59"
</code></pre>

## 🌍  Your Environment
**Deployment Environment:**
3.1.0
**EdgeX Version [**REQUIRED**]:**

**Anything else relevant?**

cloudxxx8, 2023-11-22:
you marked this as a regression bug
does this feature work in the older version of device-onvif-camera?

watjt, 2023-11-23:
> you marked this as a regression bug does this feature work in the older version of device-onvif-camera?

I resolved this issue by modifying the version of https://github.com/IOTechSystems/onvif to v1.1.1.

closed 2023-11-23 by watjt