edgexfoundry/device-onvif-camera #432: DiscoverySubnets are empty!"

outbackdingo, 2024-03-27:
Hi we started looking at edgex again to see if its come any further for onvif, previously, the hkvision camera we had weas discovered with jakarta some noths ago, then we dropped the project, well its back now however with napa we get the following error. and tyhe camera is not found. We used tui-generator for a simple onvif and mqtt exporter,  the following docker compose was created

```name: edgex
services:
  app-http-export:
    container_name: edgex-app-http-export
    depends_on:
      consul:
        condition: service_started
        required: true
      core-data:
        condition: service_started
        required: true
    environment:
      EDGEX_PROFILE: http-export
      EDGEX_SECURITY_SECRET_STORE: "false"
      SERVICE_HOST: edgex-app-http-export
      WRITABLE_LOGLEVEL: INFO
      WRITABLE_PIPELINE_FUNCTIONS_HTTPEXPORT_PARAMETERS_URL: http://EXPORT_HOST_PLACE_HOLDER:7770
    hostname: edgex-app-http-export
    image: edgexfoundry/app-service-configurable:3.1.0
    networks:
      edgex-network: null
    ports:
      - mode: ingress
        host_ip: 127.0.0.1
        target: 59704
        published: "59704"
        protocol: tcp
    read_only: true
    restart: always
    security_opt:
      - no-new-privileges:true
    user: 2002:2001
    volumes:
      - type: bind
        source: /etc/localtime
        target: /etc/localtime
        read_only: true
        bind:
          create_host_path: true
  app-metrics-influxdb:
    container_name: edgex-app-metrics-influxdb
    depends_on:
      consul:
        condition: service_started
        required: true
      core-data:
        condition: service_started
        required: true
    environment:
      EDGEX_PROFILE: metrics-influxdb
      EDGEX_SECURITY_SECRET_STORE: "false"
      SERVICE_HOST: edgex-app-metrics-influxdb
    hostname: edgex-app-metrics-influxdb
    image: edgexfoundry/app-service-configurable:3.1.0
    networks:
      edgex-network: null
    ports:
      - mode: ingress
        host_ip: 127.0.0.1
        target: 59707
        published: "59707"
        protocol: tcp
    read_only: true
    restart: always
    security_opt:
      - no-new-privileges:true
    user: 2002:2001
    volumes:
[... 574 lines cut ...]
  consul-data:
    name: edgex_consul-data
  db-data:
    name: edgex_db-data
  kuiper-data:
    name: edgex_kuiper-data
  kuiper-etc:
    name: edgex_kuiper-etc
  kuiper-log:
    name: edgex_kuiper-log
  kuiper-plugins:
    name: edgex_kuiper-plugins
```

it comes up fine, yet no cameras are found there are 2 on the network, we checked the container logs

```
level=INFO ts=2024-03-27T04:35:01.803920868Z app=device-onvif-camera source=checkstatuses.go:186 msg="Starting task loop."
level=INFO ts=2024-03-27T04:35:01.803945128Z app=device-onvif-camera source=autodiscovery.go:46 msg="Starting auto-discovery with duration 1h0m0s"
level=INFO ts=2024-03-27T04:35:01.803968779Z app=device-onvif-camera source=driver.go:290 msg="Discover was called."
level=INFO ts=2024-03-27T04:35:02.804848982Z app=device-onvif-camera source=driver.go:334 msg="Discovered 0 device(s) in 1.000853586s via multicast."
level=WARN ts=2024-03-27T04:35:02.804900732Z app=device-onvif-camera source=driver.go:352 msg="netscan discovery was called, but DiscoverySubnets are empty!"
```

simply put we want to utilize edge to discover onvif devices in the environment and export their feeds via mqtt to mainflux, any help is appreciated

jiekechoo, 2024-03-27:
@outbackdingo Did you set the discovery network subnets before starting discovery action?

You could put the configuration to your compose file as shown below or use linux `env` settings,  and try again:
```
            # Host Network Interface, IP, Subnet
            APPCUSTOM_DISCOVERYETHERNETINTERFACE: wlp1s0 # determine this setting for your machine
            APPCUSTOM_DISCOVERYSUBNETS: 192.168.93.0/24 # determine this setting for your machine
            APPCUSTOM_DISCOVERYMODE: multicast
```
or use the `device-onvif-camera` api: https://docs.edgexfoundry.org/3.1/microservices/device/services/device-onvif-camera/swagger/ to do that.

the `auto-discovery` documentation is here: 
https://docs.edgexfoundry.org/3.1/microservices/device/services/device-onvif-camera/supplementary-info/auto-discovery/

outbackdingo, 2024-03-27:
Ive added the follwing to the docker-compose.yml, tore it all down and docker compose up -d and still same results.

    environment:
      DEVICE_DISCOVERY_ENABLED: "true"
      DEVICE_DISCOVERY_INTERVAL: "1h"
      APPCUSTOM_DISCOVERYETHERNETINTERFACE: "eno1"
      APPCUSTOM_DISCOVERYMODE: "both" # netscan, multicast, or both
      APPCUSTOM_DISCOVERYSUBNETS: "192.168.10.0/24"
      EDGEX_SECURITY_SECRET_STORE: "false"
      SERVICE_HOST: edgex-device-onvif-camera

jiekechoo, 2024-03-27:
Try to call the api manually?

outbackdingo, 2024-03-27:
ok so... 

resulted in...

```
level=WARN ts=2024-03-27T07:20:44.742714349Z app=device-onvif-camera source=discover.go:62 msg="No valid CIDRs provided, unable to scan for devices."
level=INFO ts=2024-03-27T07:20:44.742729703Z app=device-onvif-camera source=driver.go:375 msg="Discovered 0 device(s) in 68.749µs via netscan."
level=INFO ts=2024-03-27T07:20:57.880136327Z app=device-onvif-camera source=config.go:684 msg="Updated custom configuration 'AppCustom' has been received from the Configuration Provider"
level=INFO ts=2024-03-27T07:20:57.880191148Z app=device-onvif-camera source=driver.go:437 msg="Discover configuration has changed! Discovery will be triggered momentarily."
level=INFO ts=2024-03-27T07:21:07.88389168Z app=device-onvif-camera source=driver.go:290 msg="Discover was called."
level=INFO ts=2024-03-27T07:21:08.885384889Z app=device-onvif-camera source=driver.go:334 msg="Discovered 0 device(s) in 1.001415384s via multicast."
Onvif WS-Discovery: Find Xaddr: 192.168.10.127            EndpointRefAddress: a34ac000-711d-11b2-806a-240f9bb602d6
level=WARN ts=2024-03-27T07:21:10.984079267Z app=device-onvif-camera source=credentials.go:105 msg="Device a34ac000-711d-11b2-806a-240f9bb602d6 is missing MAC Address, using default secret name"
level=WARN ts=2024-03-27T07:21:11.021271903Z app=device-onvif-camera source=onvifdiscovery.go:122 msg="failed to get the device information for the camera a34ac000-711d-11b2-806a-240f9bb602d6, invalid request for the function 'GetDeviceInformation' of web service 'Device'. Onvif error: fault reason: The action requested requires authorization and the sender is not authorized, fault detail: , fault code: env:Sender ter:NotAuthorized "
level=INFO ts=2024-03-27T07:21:11.021323254Z app=device-onvif-camera source=driver.go:375 msg="Discovered 1 device(s) in 2.135874566s via netscan."
level=INFO ts=2024-03-27T07:21:11.021393003Z app=device-onvif-camera source=async.go:105 msg="Adding discovered device unknown_unknown_a34ac000-711d-11b2-806a-240f9bb602d6 to Metadata"
level=INFO ts=2024-03-27T07:21:11.031998003Z app=device-onvif-camera source=devices.go:78 msg="LastConnected-unknown_unknown_a34ac000-711d-11b2-806a-240f9bb602d6 metric has been registered and will be reported (if enabled)"
level=WARN ts=2024-03-27T07:21:11.032033469Z app=device-onvif-camera source=credentials.go:105 msg="Device unknown_unknown_a34ac000-711d-11b2-806a-240f9bb602d6 is missing MAC Address, using default secret name"
```

outbackdingo, 2024-03-27:
Okay and how/where do i set the secret name .. i dont recall having this error last time

level=WARN ts=2024-03-27T07:21:10.984079267Z app=device-onvif-camera source=credentials.go:105 msg="Device a34ac000-711d-11b2-806a-240f9bb602d6 is missing MAC Address, using default secret name"

jiekechoo, 2024-03-27:
Yes, it works. 

You should set the `secret store`, and bind the mac address to the secret name. see here: https://docs.edgexfoundry.org/3.1/microservices/device/services/device-onvif-camera/Walkthrough/deployment/ `Map credentials to devices.` section.

outbackdingo, 2024-03-27:
ok i have 1 cameras working and can show rtsp stream url in the console all try / set appear to work, though im perplexed

StreamUri is showing
```
ResponseRaw:
[
   {
      "id": "44d60484-2c79-4b93-87da-c8413890d992",
      "origin": 1711528010115316700,
      "deviceName": "IPCAM-C6F0SpZ3N0PmL2-5f5a69c2-e0ae-504f-829b-00FADE9A73A7",
      "resourceName": "StreamUri",
      "profileName": "onvif-camera",
      "valueType": "Object",
      "value": "",
      "objectValue": {
         "MediaUri": {
            "InvalidAfterConnect": true,
            "InvalidAfterReboot": true,
            "Timeout": "PT5S",
            "Uri": "rtsp://192.168.10.126:554/11"
         }
      }
   }
]```

but that is the cameras interanl ip, so how do i export the cameera stream to mqtt to mainflux... or is this not feasible?

i would have thought streamUri was the embedded device running edgeX .....

i simply want to discover devices on the network and export their rtsp streams via mqtt to the cloud....

the docs seem vague on how to accomplish this? is it even feasible?

jiekechoo, 2024-03-27:
> but that is the cameras interanl ip, so how do i export the cameera stream to mqtt to mainflux...

Yes, it can be.
Here is a simply guide you can refer:
- use a `AutoEvent` command for getting the `StreamUri` readings to the EdgeX MessageBus;
- setting a mqtt export to send the readings to the mqtt broker;
- Optional: transfer the mqtt message to mainflux..., it depends on your case's env.