edgexfoundry/device-onvif-camera #434: Failed to Get profiles of Camera Device that have been successfully deployed on EdgexFoudary ! 404 Error!

find-think-solve, 2024-05-10:
🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑
curl -s http://0.0.0.0:59882/api/v3/device/name/Pangu/MediaProfiles | jq -r '"profileToken: " + '.event.readings[].objectValue.Profiles[].Token''
jq: error (at <stdin>:1): Cannot iterate over null (null)
🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑🛑
I have successfully deployed the Pangu device, and tried the core command in the UI as follows:

[image]
However, when i try to get the profiles of Pangu, it turned to be an error.  And i discard the requirement of token, i returned with an 404 error.  
| curl -s http://0.0.0.0:59882/api/v3/device/name/Pangu/MediaProfiles | jq .
{
  "apiVersion": "v3",
  "message": "request failed, status code: 404, err: {\"apiVersion\":\"v3\",\"message\":\"fail to query device by name Pangu\",\"statusCode\":404}\n",
  "statusCode": 404
}
I have tried to restart docker and reconfig internet, but get no result.
This issue has been bothering me for a long time，help and support are expected.  I express my sincere gratitude for any possible solutions.

jiekechoo, 2024-05-12:
@find-think-solve 
Trying this url get all devices:
```
curl -s curl -s http://0.0.0.0:59882/api/v3/device/all | json_pp 
```
and, can you find the device named `Pangu` ?

find-think-solve, 2024-05-12:
> @find-think-solve Trying this url get all devices:
> 
> ```
> curl -s curl -s http://0.0.0.0:59882/api/v3/device/all | json_pp 
> ```
> 
> and, can you find the device named `Pangu` ?

Thank you for your attention, you are so kind😊. Before that I reinstalled the virtual machine system and the problem no longer occurred😅. My guess is that I accidentally changed the network configuration of the VM or docker. 
Now I meet a new problem.  It seems that the app-edge-camera-management docker image is lost and i did not find the image in docker hub. 
[image]

find-think-solve, 2024-05-12:
[image]
[image]

jiekechoo, 2024-05-13:
You should build the docker image `app-camera-management:0.0.0-dev` by yourself.

> ![service](https://private-user-images.githubusercontent.com/57736570/329883185-d75af2f7-8cfc-43c2-a580-177d15d683c0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU1NTk1MzQsIm5iZiI6MTcxNTU1OTIzNCwicGF0aCI6Ii81NzczNjU3MC8zMjk4ODMxODUtZDc1YWYyZjctOGNmYy00M2MyLWE1ODAtMTc3ZDE1ZDY4M2MwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDAwMTM1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWExYzY0MjIxNTE1MjI1MTBmNDc1N2MzOWVmY2RjOWRkMjA2YmY0OTlkZTIyNDEyYmMwZjI0ODgxMDNhODdmMGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-fYoB4gVdzGO8XNeLrQjqs1pS0RZ_v-KA96c6Pi464M) ![docker](https://private-user-images.githubusercontent.com/57736570/329883192-3a4fef38-b700-4c5a-9336-5e989e998292.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU1NTk1MzQsIm5iZiI6MTcxNTU1OTIzNCwicGF0aCI6Ii81NzczNjU3MC8zMjk4ODMxOTItM2E0ZmVmMzgtYjcwMC00YzVhLTkzMzYtNWU5ODllOTk4MjkyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDAwMTM1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQxNWE5MTQ2OTE0ODQzM2FmMzIwZWM5MjFmYmY5NWZjNTJmNjM4NmU0OTU2OGI2ZmZiMDIzZjZiYjM2ZmYyZTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2xpmlVwVK-kUxf1mHxnXpJnFDFDiJHzrCpupikpm690)

find-think-solve, 2024-05-13:
> You should build the docker image `app-camera-management:0.0.0-dev` by yourself.

Thank you for your  suggestion sincerely
[image]
. I want to give it a try according to your suggestion. But the picture is invalid, could you re-upload the reference picture? I can’t see the picture here.
[image]
. I want to give it a try according to your suggestion. Could you re-upload the reference picture? I can’t see the picture here.
[image]

jiekechoo, 2024-05-13:
It's your picture that I just quote it.
The application service named `app-camera-management`, you should build it to docker image `app-camera-management:0.0.0-dev` in your host environment. https://docs.edgexfoundry.org/3.1/examples/app-service-examples/camera-management/Ch-CameraManagement/#2-start-edge-video-analytics-microservice-running-for-inference

find-think-solve, 2024-05-13:
> It's your picture that I just quote it. The application service named `app-camera-management`, you should build it to docker image `app-camera-management:0.0.0-dev` in your host environment. [...]

Okay, I'll study the tutorial carefully. If there is any latest progress, I will give a feedback as soon as possible.

find-think-solve, 2024-05-14:
> It's your picture that I just quote it. The application service named `app-camera-management`, you should build it to docker image `app-camera-management:0.0.0-dev` in your host environment. [...]

Exciting news : I have successfully established the app according to your suggestion, and started the relevant container.😊
However, I visit localhost : 59750, it returned with ' message not found '.  localhost:4000 is normal and "get" command of my Camera001  in the ui is valid.   
[image]
[image]
[image]

find-think-solve, 2024-05-14:
@jiekechoo 
the situation of building app is shown in the picture, could you help me find the cause of Not Found. Thank you for your many guidance sincerely☺. 
[image]