edgexfoundry/device-onvif-camera #485: An error is reported when accessing the onvif-camera

ccxyChuzhong, 2024-12-17:
The IP address is correct, and the port is correct. However, an error is reported when obtaining the stream address: code: 500, message: request failed, status code: 500, err: {"apiVersion":"v3","message":"error reading Regex DeviceResource(s) SnapshotUri for Huawei Camera - 153 -> failed to execute read command -> failed to create 'GetSnapshotUri' response for the web service 'Media' -> EOF","statusCode":500}
[image]

ccxyChuzhong, 2024-12-17:
What's going on, please? How to deal with it?

ccxyChuzhong, 2024-12-17:
How do I configure my credentials if they are not configured?

ccxyChuzhong, 2024-12-17:
Get the profile token by executing the command: GetProfiles
[image]

ccxyChuzhong, 2024-12-17:
[image]

cloudxxx8, 2025-01-14:
please provide the full steps and service logs.  there is no clue what you are trying to do here.
you can also try to use the generic SOAP client to invoke the camera and see if it works normally.